PX4/PX4-Autopilot #1986: Add support to DISTANCE_SENSOR msg

TSC21, 2015-04-04:
The idea is to have a topic than can subscribe to external sensors connected to a offboard computer and give the option to use those instead of the ones connected directly to the FCU. For example, as the LIDAR Lite may be connected through I2C/PWM, we can also have multiple sensors connected to an offboard computer which information may be sent through a serial connection to the FCU (p.e. using MAVROS).

TSC21, 2015-04-04:
Some ideas on how to add this support are welcomed! Thanks!

blankered, 2015-04-05:
I second this idea, I would really would like to figure this out.

mhkabir, 2015-04-05:
Why would one connect external low level peripherals to the companion computer? Use whatever bus you want to on the Pixhawk. 
Also,  there is a distance sensor uORB topic. The EKF currently uses this topic. Add a mavlink publisher for this and you can get the data offboard.

TSC21, 2015-04-05:
@mhkabir you are certainly right ;) but that does not mean it can't be an addiction to the firmware as an alternative to the main buses, which can suffer from some kind of overload when many sensors are connected (it may also solve the I2C problem of legacy LIDARLite, as an alternative to using PWM pins).
Relative to the EKF part, yes I saw that there is a distance sensor topic, but for external communication, from what I know, we need a message stream/publisher for that, which means adding support for `DISTANCE_SENSOR`.
I'll be up for this in the next days. Any other comments and ideas are welcomed :)

andre-nguyen, 2015-04-05:
I could see this feature maybe being used with this: http://wiki.ros.org/laser_height_estimation

TSC21, 2015-04-05:
Nice catch @andre-nguyen! As soon as I launch a PR, I'll let you know!

TSC21, 2015-05-13:
The PR in mavros side to support this message has been launched: https://github.com/mavlink/mavros/pull/292. If someone want to give some ideas, feel free and thank you :)

Btw, regarding the `covariance` value: shouldn't this be `variance` instead, since it's a one-dimension measurement related, i.e., it's relative to `z` only and not other dimensions?

TSC21, 2015-05-16:
The other thing I want to ask is the following:
https://github.com/mavlink/c_library/blob/master/common/mavlink_msg_distance_sensor.h#L13 where is this enum? Does it exists at all?
And related to this:
does https://github.com/mavlink/c_library/blob/master/common/mavlink_msg_distance_sensor.h#L12 has any relation with the sensor orientation?
And again: shouldn't it be `variance` instead of `covariance`.
@LorenzMeier can you clarify this please? Thank you!

TSC21, 2015-05-16:
Oh, and just another thing:
Is it possible to add to `MAV_DISTANCE_SENSOR` enum `MAV_DISTANCE_SENSOR_INFRARED`? (p.e., for short range infrared rangefinders)

LorenzMeier, 2015-05-17:
> I have this doubt:
> https://github.com/mavlink/c_library/blob/master/common/mavlink_msg_distance_sensor.h#L13 where is this enum? Does it exists at all?

In common.xml, if not, please send a PR to add it.

> And related to this:
> does the sensor id (https://github.com/mavlink/c_library/blob/master/common/mavlink_msg_distance_sensor.h#L12) has to have any relation with the sensor orientation?

We still have to add that ENUM. Probably something in 45 degree steps like APM and PX4 use for their internal sensor config structures.

> And again: shouldn't it be variance instead of covariance?

The covariance of a vector of observations is the variance (in general). So the terminology is correct, maybe a bit confusing.

TSC21, 2015-05-20:
https://github.com/PX4/Firmware/pull/2200 is the solution for this issue. Will close this now, has it's better to continue any discussion on the PR comment section.

closed 2015-05-20 by TSC21