PX4/PX4-Autopilot #2001: Px4Flow not running while not powered from USB.

h46incon, 2015-04-09:
If my pixhawk powered from USB, it can switch to POSCTL with px4flow indoor. And flow data will be shown in QGC. Then if I keep it power up, plug battery, unplug from USB, px4flow can still work fine.

But if pixhawk powered from battery when booting, the px4flow will not working. I have added some log, I found `local_position.xy_valid` and `local_eph_good` is alway false:
```
# commander.cpp
# line 1444
check_valid(local_position.timestamp, POSITION_TIMEOUT, local_position.xy_valid
	 && local_eph_good, &(status.condition_local_position_valid), &status_changed);
```
 It seems  that flow position estimator is not running. 

So if it is a small bug, or it is just a test to run px4flow when powered from USB?

BTW:  when will the script `/etc/init.d/rc.usb` execute in startup? I find some codes releated to px4flow:
```
mavlink stream -d /dev/ttyACM0 -s OPTICAL_FLOW_RAD -r 10
...
mavlink stream -d /dev/ttyACM0 -s LOCAL_POSITION_NED -r 30
```

h46incon, 2015-04-10:
I connected nsh and read the start log, and I found that if pixhawk powered from USB, px4flow will start sucessfully:
```
[PX4FLOW] on I2C bus 1 at 0x42 (bus: 100 KHz, max: 400 KHz)
```

But if powered from battery, it will failed:
```
px4flow: driver start failed
```
However, I can start px4flow driver manually:
```
nsh> px4flow info
px4flow: driver not running
nsh> px4flow start
[PX4FLOW] on I2C bus 1 at 0x42 (bus: 100 KHz, max: 400 KHz)
```
And after that, I can switch in POSCTL mode indoor just like powered from USB when booting.

h46incon, 2015-04-10:
If I try to start px4flow after few seconds, it will works. I put these lines in `/etc/extras.txt`:
```
#!nsh

# start px4flow
echo "[px4flow] delay for px4flow ready"
sleep 10
echo "[px4flow] delay finish"
px4flow start
```
And px4flow driver will start successfully.

LorenzMeier, 2015-04-10:
Are you powering flow via USB?

h46incon, 2015-04-10:
No, flow is connected to pixhawk via I2C.

mlstolk, 2015-04-15:
I've got the exact same problem over here: booting while on USB starts the PX4FLOW (i2c) and it keeps running when I plug the batteries in and remove the USB connection. When the battery is plugged in first, it does not (auto) start.
Furthermore the PX4FLOW's data (distance / quality) seem not to be ending up in my log files on SD?

JessieHIT, 2015-04-20:
I also have this problem. When I boot the pixhawk with battery, px4flow will not work whether it is connected with pixhawk through i2c or usart.

davids5, 2015-04-20:
I have a unit on order. As soon as it arrives I will have a look at this issue.

davids5, 2015-05-19:
@JessieHIT  @h46incon @mlstolk I am consolidating the issue. Please join [this] (https://github.com/PX4/Firmware/issues/2145) discussion.

closed 2015-05-19 by davids5

davids5, 2015-05-19:
[[SOLVED] sort of] (https://github.com/PX4/Firmware/issues/2145)