PX4/PX4-Autopilot #2004: sf0x doesnt work anymore in master for LightWare SF02/F Laser Rangefinder

severinleuenberger, 2015-04-09:
As widely discussed here
https://groups.google.com/forum/?hl=de#!topic/px4users/ZQ4d6HCDLSs
the actual sf0x driver for the SF02/F Laser Rangerfinder doesnt work anymore in the actual master. 
Could you please look at this?
Best wishes, Severin

bansiesta, 2015-04-12:
I think the problem could stem from a problem if you supply a custom port:

First, the custom port is opened here:
https://github.com/PX4/Firmware/blob/master/src/drivers/sf0x/sf0x.cpp#L203

It is then probably later in the init() call closed.
https://github.com/PX4/Firmware/blob/master/src/drivers/sf0x/sf0x.cpp#L280

And then opened again but this time the default port instead of the custom one:
https://github.com/PX4/Firmware/blob/master/src/drivers/sf0x/sf0x.cpp#L636

SimonWilks, 2015-04-13:
So @bansiesta was spot on. Before fixing this I tested on the default port (ttyS6 / Serial4 on the Pixhawk) and it worked fine. With the fix other ports (tested ttyS2) work too now. If it's still not working it will be a board specific or wiring issue.

Also make sure nothing else is trying to use the same port (ie: mavlink).

PR on its way shortly.

PR #2017 (2015-04-13): Store the port so the in use port will be referenced when reopening.
This allows the SF0x lasers to operate on non-default ports and resolves #2004. Tested on both the default port (Serial4) and Telem2:

```
nsh> sf0x start /dev/ttyS2
nsh> sf0x test
sf0x: single read
sf0x: val:  0.08 m
sf0x: time: 28345363
sf0x: read #0
sf0x: val:  0.080 m
sf0x: time: 28429739
sf0x: read #1
sf0x: val:  0.080 m
sf0x: time: 29103631
sf0x: read #2
sf0x: val:  0.080 m
sf0x: time: 29606865
sf0x: read #3
sf0x: val:  0.080 m
sf0x: time: 30109990
sf0x: read #4
sf0x: val:  0.080 m
sf0x: time: 30613180
sf0x: PASS
nsh> s [... cut]
closed 2015-04-13 by LorenzMeier