PX4/PX4-Autopilot #2007: ms5611: bug in reset code

KIrill-ka, 2015-04-10:
According to ms5611 datasheet ms5611 requires 2.8ms delay after issuing a reset command.
The driver code calls start_cycle immediately.

    if (ret != OK) {
    //log("measure error %d", ret);
    /* issue a reset command to the sensor */
    _interface->ioctl(IOCTL_RESET, dummy);
    /* reset the collection state machine and try again */
    start_cycle();
    return;
    }
In it's turn start_cycle schedules measurement cycle in 1 tick.

    work_queue(HPWORK, &_work, (worker_t)&MS5611::cycle_trampoline, this, 1);
Sometimes this causes sequence of resets until the system becomes busy enough to delay for 2+ ms.
I sniffed i2c bus with logic analyzer before finding this. For me that sequence of resets cause random failures of ArduPlane baro initialization.
I've added more delay to work_queue call above to fix the problem.
       work_queue(HPWORK, &_work, (worker_t)&MS5611::cycle_trampoline, this, USEC2TICK(2800));

LorenzMeier, 2015-04-10:
Thanks! So you are using the MS5611 on a FMUv1 or on an external bus?

KIrill-ka, 2015-04-10:
The board is called "captain 3 pro". It's quite similar to FMUv1 (same cpu, mpu6500@spi, lsm303d@spi, ms5611@i2c, AT45DB, spi microsd). It was rather easy to port ardupilot to this board.

mhkabir, 2015-04-11:
@KIrill-ka Do you have a link for your board?

KIrill-ka, 2015-04-11:
http://www.aliexpress.com/item/STM32F405-flight-control-sensor-module-MS5611-MPU6500-LSM303-axis-flight-control-module-offer-four-axis-source/32259957338.html
I didn't find where the design is originating from. It comes with CD which includes the schematic.

commit 73d179fb (2015-06-15): MS5611 driver: Fix reeset logic via I2C, minor code style fixes. Fixes #2007, identified by Kirill-ka
closed 2015-06-30 by LorenzMeier