PX4/PX4-Autopilot #2009: Radio Channel reset and can't arm?

wangsen1312, 2015-04-10:
I just update the PX4's firmware and calibrate all the sensors and radio. But every time I reboot the pixhawk, I can't arm the arducopter, it always show  ARM: THR below FS, I get and find the parameters for the Throttle became 899, it should be 1119 after my calibration. 
Can anyone give me some suggestion?

DonLakeFlyer, 2015-04-10:
Which exact parameter?

wangsen1312, 2015-04-10:
@DonLakeFlyer  The Throttle channel will reset to 889 every time when I reboot? can I hard code the parameter? Thank you very much!!

DonLakeFlyer, 2015-04-10:
Which exact parameter are you talking about? There are multiple throttle parameters.

wangsen1312, 2015-04-10:
[image]
The throttle In this picture always reset? Thank you!

DonLakeFlyer, 2015-04-10:
Sorry. You're in the wrong place. You need to be in the APM forums for Mission Planner or APM Planner. This is for users of the px4fmu firmware (not APM firmware) and or QGroundControl.

closed 2015-04-10 by DonLakeFlyer