PX4/PX4-Autopilot #2015: Validate EKF attitude estimation during hand/catapult launch

kd0aij, 2015-04-13:
[image]
This plot shows a negative-going spike of approximately 10 degrees in the pitch attitude estimate coincident with the forward acceleration of a hand launch. Logfile: http://dash.oznet.ch/view/FSCop8iEPryB6oPxsQhFqN
I don't think this pitch deviation is real, based on observation of the actual hand launch trajectory.
A catapult launch will be performed to collect additional data.

LorenzMeier, 2015-04-13:
You were flying 7f9138f400dc9efcb86fbbd44907fcda5681b1fb, which was lacking https://github.com/PX4/Firmware/commit/368a21ddab652ede39ffee3bd0a4f8e770fd0c8d, which in turn lead to a wrong strap down fusion of accelerations. I'm pretty sure if you re-do the test on the bench with master this will be gone.

kd0aij, 2015-04-13:
Thanks Lorenz. I thought I had that fix, but must have forgotten to reflash the Ninox after merging.

kd0aij, 2015-04-14:
Catapult launch test with recent master:
Firmware Version	4b34221, log: http://dash.oznet.ch/view/2tKinhfPJVz3sT4okLRuhk
[image]

I think the few degrees of upward pitch on acceleration are probably real. Video coming soon.

kd0aij, 2015-04-14:
catapult launch video: http://youtu.be/Px7PTE8WC9A

kd0aij, 2015-04-15:
[image]

This plot indicates that the EKF pitch estimate agrees well with the integral of the pitch axis gyro.
Closing.

closed 2015-04-15 by kd0aij