PX4/PX4-Autopilot #2020: Expose ESC calibration tool via MAVLink

LorenzMeier, 2015-04-13:
We have ESC calibration, but its not exposed to average users right now.

tumbili, 2015-05-13:
Support for IO but not FMU

LorenzMeier, 2015-05-15:
@DonLakeFlyer would you have the opportunity to do the required button / view while on the road? We kind of need this for the release, and I think it would fit the power config pane. It should launch a right side menu asking to clear the props, then launch.

DonLakeFlyer, 2015-05-15:
May take me a day or two to get back to Internet

DonLakeFlyer, 2015-05-19:
So the user is supposed to be connected via USB but not initially plugged in to battery? Then they start calibration. It arms and then the user is supposed to plug in battery which should power the ESC with a high throttle. Then either click or wait 5 seconds to go to low throttle. Then calibration is done and user should unplug battery.

All of the above needs to happen with the props off.

DonLakeFlyer, 2015-05-19:
The do_esc_calibration routine looks a bit not quite right? It is using err() to display errors. This means the error text won't make it to QGC, also I'm not very familiar with err() but from my search results, finding implementation in err.c, this ends up calling exit, which would seem bad? Won't this cause commander to exit? Maybe I don't understand what that does.

LorenzMeier, 2015-05-19:
@DonLakeFlyer You're right, it doesn't look quite right. It needs to use the MAVLink critical output instead. @tumbili Could you fix that in a PR?

DonLakeFlyer, 2015-05-19:
I can fix this up. I want to change the messaging to follow how the other cal text messages work as well. Then I'll do the QGC side of things.

DonLakeFlyer, 2015-05-19:
I think I spoke too soon. I'm not quite sure what to do with the error cases. I assume that if some of those ioctl's fail that you need to reset some state back as well. For example if PWM_SERVO_SET_ARM_OK succeeds but PWM_SERVO_ARM fails I would assume that some sort of ioctl thing must be sent to reset the safety states back to normal?

Also: Is the user supposed to be moving the throttle themselves? Or does the firmware send the right signals for high and low throttle? If firmware does it, where is that in the code?

LorenzMeier, 2015-05-19:
The Firmware doesn't mess with the throttle, instead it tells the actuator to move into the calibration position. From an API perspective that's much more sensible, because actuators not supporting calibration (e.g. UAVCAN) can just ignore it, instead of actually moving their throttle position.

drv_pwm_output.h has all the details you need, e.g. to reset the arm ok flag use PWM_SERVO_CLEAR_ARM_OK.

DonLakeFlyer, 2015-05-19:
Ahh, I assume actuator calibration is triggered by: armed.in_esc_calibration_mode = true? Wasn't seeing that since it's not in the calibration routine but outside in commander. I assume it would be ok to move this internal to the calibration routine?

LorenzMeier, 2015-05-19:
The armed topic gets repeatedly published by the commander, so you would need to take a reference / pointer and update it from there.

LorenzMeier, 2015-05-22:
@DonLakeFlyer anything we can assist here with?

LorenzMeier, 2015-05-23:
@DonLakeFlyer @tumbili Thanks for getting this done!

closed 2015-05-23 by LorenzMeier