PX4/PX4-Autopilot #2021: Implement ALTCLT for fixed wing.

tumbili, 2015-04-14:
Currently altitude control mode is not implemented for fixed wing planes.
In this mode the plane should keep its current altitude. The speed of the plane can be mapped to the throttle stick. The user can command climb/descend rate with the pitch stick and roll with the roll stick.

severinleuenberger, 2015-04-14:
Please let us discuss this implementation first. 
The actual ALTCTL Mode without altctl is very useful, and on our actual GALLO paper for BAZL often used as fallback-Flightmode, if something goes wrong. 
It is very helpful to have a basic flight mode, who only stabilizes "level" to some degree, because it is a robust solution. At least we would wish to have a flightmode like that. Ok, the name ALTCTL sounds like "control of altitude level", and therefore it is the proper name for an implementation like this. But we like to have the actual "ALTCTL" mode as well. Maybe you can somehow copy the code from the actual ALTCTL and provide it as a new Flightmode like "STABILIZED"? So that we can choose between those?
Best, Severin

tumbili, 2015-04-14:
@severinleuenberger Yes I agree on that. I also started on APM and I found their solution of flight modes very intuitive. I just opened the issue so that we know that there is work to be done here.
I am also not very sure myself what is working at the moment and what not. I will have to look into this.
But your suggestions very much make sense and are welcome.

thomasgubler, 2015-04-14:
We have something similar already when in POSCTL https://github.com/PX4/Firmware/blob/master/src/modules/fw_pos_control_l1/fw_pos_control_l1_main.cpp#L1254

As Severin mentioned, please don't remove attitude stabilized (fixedwing ALTCTL atm). Maybe it's time for the long overdue mode cleanup? Some history regarding the naming mess: the ALTCTL mode was first called SEATBELT and the POSCTL mode was called EASY. Then some copter guys ( :wink: ) renamed the modes without thinking of fixedwings which you can fly in *true* manual mode (MC manual is actually attitude stabilized...).

tumbili, 2015-04-14:
@thomasgubler Yes I think it's time for a mode cleanup :-D So why not introduce a STABILIZE mode just like in APM which just controls attitude. Then ALTCTL can really imply altitude control mode as the name suggests. I often use STABILIZE mode for landing aid and ALTCTL (when I used to fly APM) for people who are e.g. flying for the first time and like to have a very "safe" mode. So I think we should implement both. The commander can treat STABLIZE like MANUAL if the vehicle is a rotary wing.
Any comments?

thomasgubler, 2015-04-14:
We would need to make sure that all output in qgc is correct. I am not sure how much of it is hardcoded in QGC. Also the RC stick mapping is going to differ between MC and FW,

AndreasAntener, 2015-04-15:
Sounds great :). I also agree with you on the modes: attitude stabilized, altitude control, position control (near future? ;)). Not sure how far you'd like to go with the mode cleanup, if it's just about fixed wing or more, e.g. #2031 (it's not really about modes, more about the state machine that handles them)

This just reminded me of something else: #2030

tubeme, 2015-05-26:
It is VERY VERY important to clear all the mess with the MODE NAMES for the MC and Plane modes. There is one huge change made between APM and PX4 (I dont know when it happened) but as a consequence there is 10 times bigger mess in the forums and internet. Google indexed all kinds of modes with different names as a different idioms, despite they are the same thing people are talking about. I saw people on forums spending a lot of time to figure out that they are talking about the same thing or parameter not knowing that it is the same function, just because one uses the old convention and the other uses the new convention of names. Peopla are still talking about FBW-A and FBW-B... So my suggestion is think twice and made the change once and at least for some fixed amount of time. Let say for the next 3 years we are going to use "this and this and this names". We are flying Pixhawk in a couple of multirotors and in couple of plains and there is little to no sense using the ALTCTL and POSCTL for planes. I think that who came up with the MODE NAMES comes from multicopter world! It is obvious. ALTCTL and POSCTL are inadequate for Planes. Also when I build planes and want to search something on the net with key ALTCTL i find zillion results for multi-copter and not planes. Google does not understand English nor names so it cannot differentiate between the vehicles and modes we are talking about, thus finding the right information for me, sometimes takes days and weeks. And we should agree that in our business Internet is our helper, teacher, users manual, customer support and problem solver. Pixhawk was the most confusing FC to set up for me so far and I'm quite experiensed user with CompSci education. Because the Planes and Multicopters are totally different in terms of way of flying it is very very good to make totally different names for the two modes!!! This way we will understand each other better and we will be able to find easier what we are looking for in the net. When we use STABALIZE for the two vehicles they mean different thing as well as MANUAL as well as LOITER as well as ALTCTL and POSCTL. PosCTL is actually GPS hold from DJI convention. ALTCTL is totally inadequate mode for the MC world I and nobody I know use it ever!!! I would love to have Loiter for the MC pointing the front towards a centered spot. I would call this mode CIRCLE for MC. The same is with RTL it is rather RTH because nobody I know commands the machine to auto land in failsafe. Everybody uses it to RTH and then gain control and land the machine itself. Of course if you dont gain control the machine should land as DJI does besides their mode is RTH. You call the spot a HOME spot not a LAND spot even in QGroundControl it is the HOME position not land position. You can land the machine wherever, but there is just one position that the machine took off and it is its current HOME. Manual in MC is really a stabalized flight. In the Planes world it is different story. I can go on for the planes modes in the same way. The only good name for the planes is LOITER! And the only good adequate name for both is MISSION. There is a lot of marketing in making the correct names sound and mean CORRECT things. And this is part of the success of DJI. It is not only the good FC that works great. It is the way it is used and the way we communicate about it. I must say I'm surprised by the BETTER performance of Pixhawk over DJI!!! But if I did not have a lot of knowledge about FC's I couldn't set it up nor I could have found my way in the forums. QGROUNDCONTROL should probably have two flavours for begginers and for advanced users. I had to hardware re-solder my servos in terms to reverse them because I could not find a way to reverse the logic in the MIXER... and in ALTCTL the FC gave wrong commands to the servos... The mixer files are hard coded so in this instance I used the Bixler profile  and had to re-solder AILERONS and RUDDER servo... There was no custom mixer for the planes mode!!!

tumbili, 2015-05-27:
@tubeme Hi, thanks for the feedback. You have to keep in mind that PX4 does not strictly differentiate between fixed wing and multicopter modes. You could have these completely separate but then you will also duplicate alot of code. I agree that the naming of the modes lead to a lot of confusion in the bast. This is why we are working hard to correctly implement the modes. There is an issue which deals with the reversal of the pwm outputs https://github.com/PX4/Firmware/issues/2215

tubeme, 2015-05-27:
I understand what you say and I agree. I just had my point. The future will show. I just does not understand how is it the same in ALTCTL when you have centered throttle in MC and 0 throttle in Planes mode? Or in Manual you have stabalized flight in MC vs manual control in Planes mode.

closed 2015-05-31 by LorenzMeier