PX4/PX4-Autopilot #2027: Geofence test flights

LorenzMeier, 2015-04-14:
@thomasgubler Does the test plan below make sense?

@kd0aij When you go flying next time a geofence test would be awesome. The parameters are explained here:
http://www.pixhawk.org/firmware/parameters#geofence

  * GF_MODE to 2
  * GF_MAX_HOR_DIST to 20
  * GF_MAX_VER_DIST to 10

This should trigger an RTL when flying out more than 20m or flying higher than 10.

dagar, 2015-04-14:
I finally tested this outside and it works, but someone else verifying couldn't hurt. I need to limit the mavlink critical messages. I must have been muted before during HIL testing. PR #2028

Currently the geofence distance violation just warns over mavlink. I'm working on RTL right now.

LorenzMeier, 2016-02-18:
Closing, stale!

closed 2016-02-18 by LorenzMeier