PX4/PX4-Autopilot #2030: Altitude hold (currently POSCTL) should prioritize altitude over speed

AndreasAntener, 2015-04-15:
As I recall, the last time I tested POSCTL for fixedwing I had the issue that commanded speed had priority over commanded altitude. So if it was not able to hold the speed it would start to dive. IMHO that's wrong, priority should be to hold the altitude.

AndreasAntener, 2015-04-15:
@friekopter @thomasgubler not sure if someone looked into this already, maybe I'm not up-to-date

LorenzMeier, 2015-04-15:
@AndreasAntener You have a point there, I'm just not sure yet how to do this properly. What you are proposing is to essentially risk to stall, which doesn't work either. It sounds like we would need to find an approach which changes the speed / altitude weight (which we already have!) based on airspeed. Or we just rely on the under speed protection kicking in, which might in fact be all it needs.

AndreasAntener, 2015-04-15:
Yes, I'm counting on the underspeed protection to kick in if necessary. The issues I had were related to commanding a speed that the airframe just couldn't do (on level flight). The same can happen if you climb too steep. But as long as the speed doesn't go below minimum, I don't see any reason to limit it's climbing or go into a dive (however during a climb I can understand if people would want it the other way around).

This speed/altitude weight you're talking about.. Somehow I heard that before from Thomas too. I think back then we just talked about tweaking that, but it never got so far. That's something in TECS no? Maybe we should just try that then?

LorenzMeier, 2015-04-15:
TECS params are documented here:
http://www.pixhawk.org/firmware/parameters#fixed_wing_tecs

Look for ```FW_T_SPDWEIGHT``` and play with it on your next flight.

friekopter, 2015-04-15:
I am not sure what the requirement is in the different situation. As FW_T_SPDWEIGHT is 1 by default it should hold an equal tradeoff between speed and altitude. 
What I observe in flight tests:

If I command maximum sink rate and minimum speed the autopilot constrains the real sink rate in order to not exceed commanded speed too much.  I have to check that in my logs, but that is what I observe. 
Same in max climb rate. 
It will not dive until it comes close to stall speed, maybe due to turbulences or bad configuration parameters.(I have a "stall" call added in my audio)

I think that is a reasonable reaction because a stall must be avoided with highest priority, and inside the flight envelope the autopilot can not really make a decission what is more important speed or climb rate (Would like to know how that is handled in real aviation)..
To come to the situation you describe in the first place: I agree: As soon as climb rate is set to 0 it does not look very smart to leave altitude just to reach an unreachable speed, as long minimum speed is reached. 
On the other hand: If the parameters are not set correctly I can not expect a perfect reaction. But, again,  there may be situation where the max speed parameter is set correctly, but for some reason (low battery, gusts) speed is not reached, at least temporary. This may lead to a poor altitude hold. 

So for this case it might be a solution to overweight altitude hold by a factor proportional to 
<code>(|maxClimbRate| - |commanded climb rate|)/|maxClimbRate|</code>

friekopter, 2015-04-16:
i.e. 
<code>FW_T_SPDWEIGHT = 1 - OVERWIGHT_ALTITUDE_FACTOR * (|maxClimbRate| - |commanded climb rate|)/|maxClimbRate|</code>

and 0 < OVERWIGHT_ALTITUDE_FACTOR <= 1

AndreasAntener, 2015-04-17:
@friekopter 
>  It will not dive until it comes close to stall speed, maybe due to turbulences or bad configuration parameters.(I have a "stall" call added in my audio)

What you're saying makes sense, but I don't think it behaves like this ATM. I fist ran into this when I had a wrong configuration for max speed (set to double the speed it could do). So it flew nicely on low throttle, but as soon as I throttled up (over 50%), it started to dive. It was never near min/stall speed and therefore should not dive. The same goes for your observation. If you command a really slow speed and start to dive, as long as your not at max speed it should not limit the descent.

It really comes down to the question if your speed is borderline (max or min) or not. E.g. the speed weight could be very low by default and increase if your going near max or min speed. I also like your idea to take the climb rate into account. Maybe we can combine the two, it would give us a certain flexibility on the style of flying you want to do. E.g. for me, I don't think it has to react on a speed loss when I'm climbing (as long it doesn't reach min speed of course).

AndreasAntener, 2015-04-17:
something like that (I'm sure someone would probably find a nicer way to calculate `weight` ;)
```
rel_speed = (speed - min) / (max - min)
weight = 0 // new parameter
if rel_speed <= 0.1 // new parameter
    weight = 10 * rel_speed
if rel_speed >= 0.9 // new parameter
    weight = 10 * (rel_speed - 0.9)

 FW_T_SPDWEIGHT = weight * (
    1 - OVERWIGHT_ALTITUDE_FACTOR * (|maxClimbRate| - |commanded climb rate|)/|maxClimbRate|
)
```

severinleuenberger, 2015-04-30:
I recognized on real condition flights with posctl on stormy front winds a heavy dive/descend. I was only able to get control of my aircraft again by switching to altctl. 
I hope this would be corrected as well with your fix.

friekopter, 2015-05-01:
@severinleuenberger would be helpful to get a logfile from your flight.

severinleuenberger, 2015-05-02:
I just come from the field where I tested POSCTL with latest dev build flashed. Everything seems to work fine.

AndreasAntener, 2015-05-24:
I finally got a chance to do some fixed wing flights again :)

Tried speed-weight 0 and 1. General feedback: I couldn't really tell a difference in "normal" conditions (5m/s wind excl. gusts) with my FX-61 where I have correct parameters for the airframe (speed max/min/trim). I think I need to do a test where I have more space to observe speed over a greater distance.

Something that I was actually wondering from the beginning with this is, why my turn radius is larger than usual in this mode. This is clearly visible on the map plots from my comparison flights (2). In stabilized I'm adding pitch in turns of course, but AUTO is also doing nice sharp turns.

(1) Weight 0:
http://dash.oznet.ch/view/Bo3pfgwnHuNTvhBNhFNisE

(2) Weight 1, comparison mission/altctl/posctl:
http://dash.oznet.ch/view/HNivaDL32XxNApqLZNf7DW
http://dash.oznet.ch/view/kJAzXQnh25z3Pwy9DGpA2C

LorenzMeier, 2015-06-12:
@AndreasAntener What did you find out about the weights - any subjective difference?

AndreasAntener, 2015-06-12:
In my recent flights I didn't pay attention to this, will try to do that the next time when I test posctl again ( #2299)

LorenzMeier, 2015-06-14:
Closing this and will track it in our other milestone issues.

closed 2015-06-14 by LorenzMeier