PX4/PX4-Autopilot #2031: RTL and LOITER should not require a manual mode switch to AUTO

AndreasAntener, 2015-04-15:
Something regarding mode dependencies is bothering me ever since: RTL and LOITER require a mode switch to AUTO. This never made sense to me (from a user perspective, of course it needs AUTO for that ;)). A while ago we thought about implementing a workaround that if you hit RTL it overrides the mode switch to AUTO. But this is probably not the best idea.

thomasgubler, 2015-04-15:
Map your switches to one channel and configure your RC as you like :) This problem stems from the hierarchical approach to the switch position parsing and the limit to 4 channels for mode switches.

AndreasAntener, 2015-04-15:
From a user perspective this makes no sense. And, on standard radios this is practically impossible: If I want a dedicated RTL switch, I need to be able to mix multiple switches into one channel and have specific PWM values assigned for certain switch state combinations. I guess OpenTX can do that (? ;)), but else..

LorenzMeier, 2015-04-19:
I think this would be fairly easy to achieve. There is no reason a single switch can't flick the system into auto and perform RTL. Anyone volunteering?

LorenzMeier, 2015-04-19:
Ok, forget it, it was too easy. PR:
https://github.com/PX4/Firmware/pull/2050

Now who is ready to test?

sjwilks, 2015-04-19:
Fantastic! Seeing we've been nagging it seems only fair that we test. We
should be out flying tomorrow afternoon.

LorenzMeier, 2015-04-22:
Fixed.

closed 2015-04-22 by LorenzMeier