PX4/PX4-Autopilot #2054: Can't start HIL: Not ready to fly: Sensors not operational

SimonWilks, 2015-04-20:
After flashing master onto my HIL Pixhawk just now I can not longer arm/fly and get the following output on the console (jMAVSim):

Init MAVLink
MSG: Switched to ON hil state
MSG: Not ready to fly: Sensors not operational.
MSG: command temporarily rejected: 176

Possibly due to commit https://github.com/PX4/Firmware/commit/7dbb6c4fa80039bd9c42a83161f18da9033ac40a?

tumbili, 2015-04-21:
@SimonWilks @LorenzMeier @Zefz
I cannot arm my Pixhawk anymore, and the big LED flashed red. The serial output says:
telem> PREFLIGHT FAIL: GYRO #0 SELFTEST FAILED
And in the driver for the gyro I found:
Note: With the exception of the self-test feature, the ST L3G4200D is
 *       also supported by this driver.

LorenzMeier, 2015-04-21:
@tumbili Gyro #0 is normally the mpu6000. What does its own status say? (mpu6000 test / info?)
@SimonWilks Yes, sorry this is due to the preflight check changes. Would you mind having a quick look at a fix? It should be relatively trivial to check for if in hil mode and force the checked flags to "good" once in.

LorenzMeier, 2015-04-21:
@SimonWilks Here you go: https://github.com/PX4/Firmware/pull/2067

LorenzMeier, 2015-04-22:
Fixed.

closed 2015-04-22 by LorenzMeier