PX4/PX4-Autopilot #2056: INAV_ENABLED not visible in QGC

SimonWilks, 2015-04-20:
The INAV_ENABLED parameter is not visible in QGC and cannot be set.

As I understand it only parameters that are referenced in code or changed will be transmitted to and hence visible in QGC. In the case of INAV_ENABLED it doesn't appear in the list of parameters and this may be because the mc_apps script first does a test on this variable which, if present and = 1 only then will start apps that actually use it.

LorenzMeier, 2015-04-25:
I can neither verify this from code inspection (it should work) nor from testing (it does work). I'm closing this, let me know if this remains after rebasing on master and using a recent QGC version.

closed 2015-04-25 by LorenzMeier

AndreasAntener, 2015-04-27:
FYI: this happened (happens) on FMUv1 because the INAV modules are commented out.

LorenzMeier, 2015-04-27:
Makes sense. They are not available anyway.