PX4/PX4-Autopilot #2065: Implement per-channel direction reversal

LorenzMeier, 2015-04-21:
Unfortunately someone in the RC industry thought its funny if servos turn clockwise or counter-clockwise, so at least for PWM outputs we need to offer a per-channel reversal parameter.

We tried to avoid this using configs, but it turns out that people use different gear even in the same airframe, so I don't know a good way around this.

kd0aij, 2015-04-21:
wasn't there some related discussion of making it easier to configure mixers to deal with things like this including trim adjustments? Currently one must edit a mixer file on the SD card to reverse a servo or change offsets, and this needs to be much more convenient to allow adjustments in the field using just an RC controller.

severinleuenberger, 2015-04-21:
Reversing chanels is nice, but a real improvement would be, if the functions, which are provided by the *.mix file (who are REALLY great by the way) could be changed  directly trough QGroundControl without removing any SD-Card. Sorry for off-toppic, but as jk0aij startet this way I felt inspired to think some steps further.
Just chaning some servo chanels does only help in a few simple fixedwing setups. 
Onyl my2cent, 
Severin

bansiesta, 2015-05-31:
I believe this is fixed by #2220, closing.

closed 2015-05-31 by bansiesta