PX4/PX4-Autopilot #2071: low memory on FMUv1

kd0aij, 2015-04-22:
recent master: Not enough memory available for 2 mavlink instances. Must use nsh console to stop mavlink on telemetry port prior to attempting to connect with QGC over USB.

LorenzMeier, 2015-04-22:
I feared that one - need to set up auto testing for FMUv1. I don't have an obvious fix to offer at this point.

kd0aij, 2015-04-22:
discovered and raised that issue on a Chromebook :)

philipoe, 2015-05-02:
Having the same issue here. Either when starting a second mavlink instances, or in our case even when just startin additional (custom = our own) drivers for external devices. Are there any plans to improve RAM consumption again to make e.g. 2 mavlink instances work on FMUv1?

LorenzMeier, 2015-07-03:
I believe recent improvements brought everything back down to operational. Ping me if not.

closed 2015-07-03 by LorenzMeier