PX4/PX4-Autopilot #2072: Cut back number of text messages

LorenzMeier, 2015-04-22:
We want the system to only spit out useful text info, not all it happens to find. This is important for situational awareness. To give a recent extreme example: The engine failure of a Quantas A380 (the engine came apart mid-flight) led to a check list of text messages which took the Co-Pilot over 20 minutes to work through. All it really was trying to say was that the engine failed, and that parts flying out of it had taken out one of the hydraulic systems.

@DonLakeFlyer This will also help with your calibration issue, although we should enforce the constraints you have also on the tech level, as discussed.

LorenzMeier, 2015-04-22:
http://en.wikipedia.org/wiki/Qantas_Flight_32

bansiesta, 2015-05-31:
Is this targeted at mavlink_log_info/critical or warnx/printf?

LorenzMeier, 2015-05-31:
Log info

LorenzMeier, 2015-06-12:
We did this pretty much already and will continue - closing this for now.

closed 2015-06-12 by LorenzMeier