PX4/PX4-Autopilot #2079: ekf_att_pos_estimator local altitude not relative to home on master

jgoppert, 2015-04-24:
I just flew with INAV_DISABLED and it looks like the altitude is absolute instead of relative.

LorenzMeier, 2015-04-24:
Could you upload a log on http://dash.oznet.ch? A recent log shows a zero-based local altitude:
http://dash.oznet.ch/view/BfkVZBTWJkQXJ56JdEEdi7

Zefz, 2015-04-24:
According to this
https://github.com/PX4/Firmware/blob/master/src/modules/ekf_att_pos_estimator/ekf_att_pos_estimator_main.cpp#L814
the altitude should be zero relative to where the first baro update is processed:
https://github.com/PX4/Firmware/blob/master/src/modules/ekf_att_pos_estimator/ekf_att_pos_estimator_main.cpp#L1405

surberj, 2015-05-29:
We still have the issue that the local altitude is not set to zero before take-off. 
In ekf_att_pos_estimator_main.cpp, ~line 840, the _local_pos_z is calculated as:
_local_pos.z = _ekf->states[9] - _baro_ref_offset - _baroAltRef;

Wouldn't it make sense if the estimator itself (states[9]) would be initialized with 0, which involves that the _baroAltRef needs to be subtracted of the measurements instead of _local_pos.z?

Plot of our last flight:
[image]
- SENS_BaroAlt starts at +340m
- _ekf->states[9] and _local_pos.z both start at -340m and are off by a very small factor (~0.02m).

LorenzMeier, 2015-05-29:
This is the global altitude in NED coordinates. The local altitude will read 0 - did you maybe have no GPS lock in that instance?

surberj, 2015-06-01:
Hi Lorenz, we have had GPS lock (between 8 and 10 satellites) and the GPS altitude starts at +410m:
[image]
[image]

LorenzMeier, 2015-06-02:
Maybe this is a plotting problem? Does the LPOS_Z altitude overlap with yellow or not?

surberj, 2015-06-02:
It is not only a plotting problem - they overlap. The LPOS_Z alone looks like:
[image]

By the way: The problem also occurs when we do not have GPS connection. It seems as if the offset _baroAltRef does not capture the "real" offset of the Baro.

LorenzMeier, 2015-06-03:
@surberj Would you have time to dive into the estimator and work out where the initialisation goes wrong?

surberj, 2015-06-03:
Yes, I'll investigate in that. I can start with it tomorrow.

surberj, 2015-06-08:
It looks as if the problem can be resolved by subtracting _baroAltRef from the baro measurements and initializing the estimator (states[9]) with zero.

I'll continue some testings on our platform and open a pull request on the repo wingtra/Firmware as soon as I am confident about the solution.

LorenzMeier, 2015-06-08:
I have a fix in the l1_fix branch. It had more considerations that needed to be taken of.

surberj, 2015-06-08:
Do you refer to the fix with the new function
void AttitudePositionEstimatorEKF::initReferencePosition(hrt_abstime timestamp,
	double lat, double lon, float gps_alt, float baro_alt)
on the branch l1_fix?

If yes, did you already test it? Do you know already when this will be merged into master? If the issue is resolved with that fix I do not have to investigate further in it I assume?

LorenzMeier, 2015-06-08:
Yes. Should be in master quite soon, depends mostly on testing bandwidth

surberj, 2015-06-08:
Ok, thanks!

LorenzMeier, 2015-06-14:
This is fixed. There is still an issue on filter resets that we need to resolve for the local position.

closed 2015-06-14 by LorenzMeier