PX4/PX4-Autopilot #2081: UAVCAN sensor configuration

pavel-kirienko, 2015-04-24:
At the moment there's no reliable way to make PX4 prefer UAVCAN-interfaced sensors over internal ones. The current implementation employs a timing hack - it waits a bit after enabling UAVCAN driver before initializing other sensors, which allows the UAVCAN driver to claim zero (main) sensor topic if an UAVCAN-interfaced sensor measurement arrives during the wait.

This is where the hack is implemented: https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rc.uavcan#L10-L13

Affected sensor types are magnetometers and barometers.

First solution that comes to mind is to add a number of configuration parameters for every supported UAVCAN sensor type, which, if enabled, will instruct the UAVCAN driver to claim main topic(s) of the sensor before first measurement arrives. Essentially, these parameters will allow the user to select whether UAVCAN sensors should be used as primary ones or not, so an appropriate name would be like `UAVCAN_PRIMARY_SENSOR_<type>`.

Another solution is to explicitly disable internal sensors via some configuration parameters, but this approach, despite its determinism, has obvious drawbacks.

Also I wonder if it makes sense to enable UAVCAN driver by default if it is present in the build. When the bus is idle, it doesn't consume any significant CPU time, so it's pretty safe in this regard. However, it takes several KB of RAM when running. APM, for instance, has it enabled by default (ignoring the fact that APM is a questionable design reference).

LorenzMeier, 2015-05-18:
There is a bus ID concept which would allow sensor consumers to query the bus ID and pick UAVCAN. We will probably do this for the 1.1 release.
https://github.com/PX4/Firmware/blob/ebc4cc3a7dfc6a705a4585c75e92969a6a0e798f/src/drivers/device/device.h#L127-L148

I'm ok to enable UAVCAN by default on Pixhawk - how many KB of RAM is it really consuming? Any room for optimisations?

Could the start handler wait for a uavcan started flag and only return once uavcan is fully initialised?
https://github.com/PX4/Firmware/blob/434520406438d8e2c7e008063933bde5fc37b7de/ROMFS/px4fmu_common/init.d/rc.uavcan#L8

pavel-kirienko, 2015-05-18:
> For now startup order seems a fair prioritisation.

OK, I guess we should leave this issue open then until the bus ID is supported by the UAVCAN driver. Do you know whether APM is going to support it as well?

> I'm ok to enable UAVCAN by default on Pixhawk - how many KB of RAM is it really consuming? Any room for optimisations?

Currently it consumes about 10k for the pool, 3k for the stack and about 4k for static objects (also it allocates an object or two in the heap, but it's a mistake and it should be fixed if we decide to make it enabled by default). With dynamic node ID allocation and firmware updates the cumulative memory consumption will probably go up to about 20~30k. This is a lot, and it should be reduced; however, at the moment I don't have a solution yet.

> Could the start handler wait for a uavcan started flag and only return once uavcan is fully initialised?

That doesn't make sense. The driver will claim a topic when the first sensor measurement arrives, not when it gets initialized. I think we can live with the current approach until the bus ID concept is implemented.

closed 2015-12-20 by pavel-kirienko