PX4/PX4-Autopilot #2084: FMUv1 logfile named sessNNN/logNNN with 3D fix available from GPS

kd0aij, 2015-04-25:
Armed only after 3D lock reported via mavlink telemetry.
This occurred twice, with the log files being named sess001/log001 and sess002/log001, respectively. The second session was generated about a minute after recovery from a GPS glitch in the first session which resulted in erratic behavior in both ALTCTL and manual modes.

commit 1da048549aee5b24861ea37d094b0d770fa9868d
logfile http://dash.oznet.ch/view/Abu5WFQQTcTTC3ocXVLHL7

probably not relevant to this issue:
This LEA6H GPS module must be bad. Here's a "glitch" from that flight which would be impossible to compensate:
[image]

LorenzMeier, 2015-05-05:
3D fix is not the same thing as time fix - it might take 1-2 minutes to get that going. Does the same happen if you let it idle 10 mins in the grass before the flight?

kd0aij, 2015-05-05:
How does the pilot know whether the system has a "time fix"?
I would consider it unreasonable to require 10 minutes idling on the ground before flight.

LorenzMeier, 2015-05-05:
I just suggested a test, and didn't imply a number. We have the time when GPS gives it to us, not earlier. And you don't need to wait for time sync to fly. If you care about the exact interval you might want to read up what the expected maximum GPS signal listen time is for a worst-case time sync interval, which would give us a good benchmark.

kd0aij, 2015-05-05:
OK, but I'm confused. As I understand it, each satellite transmits time of week continuously and the GPS driver should be receiving UTC time before acquiring any fix. What am I missing here? I have verified that gps.cpp is publishing UTC time at 5Hz (updated at 1Hz) immediately after boot.

LorenzMeier, 2015-05-17:
Could you try to reproduce this with master? We need to narrow down the symptoms.

kd0aij, 2015-05-17:
Have been flying recent master on that system but usually without a GPS
module.
Will add a GPS for future flights and look for a recurrence.

kd0aij, 2015-05-17:
have just observed using commit 66b87ac from 13 May that a log was named "sess005" which showed a 3D fix from start to finish.
Just added print statements to the code which show UTC time being published and RTC being set with fix types of 0 and 3. Since this shows RTC is valid when sdlog2 created the logfile, this might indicate a problem with the -t option in sdlog2 startup.
```
gps: gps fix type: 0                                                            
gps: gps utc time: hour: 22, minute: 52, second: 32                             
gps: setting RTC                                                                
gps: gps fix type: 0                                                            
gps: gps fix type: 3                                                            
gps: gps fix type: 3                                                            
gps: gps fix type: 3                                                            
gps: gps fix type: 3                                                            
gps: gps utc time: hour: 22, minute: 52, second: 33                             
gps: setting RTC                                                                
```

LorenzMeier, 2015-05-18:
The issue you describe exists on master, but is fixed on this branch - could you cross-test?

LorenzMeier, 2015-05-18:
This branch for testing: https://github.com/PX4/Firmware/pull/2191

LorenzMeier, 2015-05-18:
One important additional hint: On master the RTC gets reset on each Firmware upgrade. This is fixed on nuttx_next.

LorenzMeier, 2015-05-18:
tested the heck out of it, can't reproduce this.

kd0aij, 2015-05-18:
are you testing on FMUv1, and should that make any difference, since it has no backup battery? Also I see you already deleted branch #2191

LorenzMeier, 2015-05-18:
I tested on Pixhawk and it would be great if you could test master on FMUv1

kd0aij, 2015-05-18:
Tested on FMUv1. Unable to reproduce problem with GPS connected after merging latest master (using branch https://github.com/kd0aij/Firmware/tree/sdlogFileName)

closed 2015-05-18 by kd0aij