PX4/PX4-Autopilot #2086: QGC 2.3.0 (stable) and 2.4.0 (dev) cause stackfault on Pixhawk V2

SimonWilks, 2015-04-25:
Filing this bug here and not directly against QGC as the autopilot should be resilient to a broken client. 

QGC 2.3.0 and 2.4.0 (from head about 15 min. ago) both cause the PX4 stack to segfault as soon as you connect via USB (radio not tested). Power cycling and attempting to connect succeeds 2-4 times out of 10.

QGC 2.2.0 (stable) does not trigger the segfault.

Tried resetting all params (mtd erase).

Laptop OS: Ubuntu 14.10
Pixhawk: V2 (board version unknown, it came with the firefly from ETH)
Connection to Pixhawk: USB
Firmware: Master (c7ecafe99b6c6c567d9bd8ac223d5e20592644bc) and current stable flashed via QGC.

Startup output (note: script failure vtol only, tested with f330 startup and default setup as well):
```
[boot] Initialized SPI port 1 (SENSORS)
[boot] Initialized SPI port 2 (RAMTRON FRAM)
[boot] Initialized SPI port 4
[boot] Initialized SDIO
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
[i] Looking for microSD...
[i] microSD mounted: /fs/microsd
[i] Init script not found: /fs/microsd/etc/rc.txt
nshterm [6:70]
param: selected parameter default file /fs/mtd_params
[param] Loaded: /fs/mtd_params
[rgbled] on I2C bus 2 at 0x55 (bus: 100 KHz, max: 100 KHz)
nsh: pca8574: command not found
 1:   SYS_USE_IO: match
 13002: + SYS_AUTOSTART: match
+ VT_MOT_COUNT: unchanged
+ VT_IDLE_PWM_MC: unchanged
[i] Config not found: /fs/microsd/etc/config.txt
px4io: CRCs match
dataman: Power on restart
dataman: Initialized, data manager file '/fs/microsd/dataman' size is 101504 bytes
[MS5611_SPI] on SPI bus 4 at 2 (11000 KHz)
ms5611: no device on bus 4
[MS5611_SPI] on SPI bus 1 at 3 (11000 KHz)
ms5611: no device on bus 1
ms5611: no device on bus 2
<adc> init done
ver hwcmp match: PX4FMU_V2
hmc5883: no device on bus 2
hmc5883: driver start failed
hmc5883: no device on bus 1
hmc5883: driver start failed
mpu6000: driver start failed
[MPU6000] on SPI bus 1 at 4 (1000 KHz)
[L3GD20] on SPI bus 1 at 1 (11000 KHz)
[LSM303D] on SPI bus 1 at 2 (11000 KHz)
meas_airspeed: no MS4525 airspeed sensor connected
ets_airspeed: no ETS airspeed sensor connected
ets_airspeed: no ETS airspeed sensor connected
px4flow: driver start failed
sensors_task: valid configs: 2 gyros, 1 mags, 2 accels
commander: loaded mission state: dataman_id=0, count=0, current=0
sensors_task: valid configs: 2 gyros, 1 mags, 2 accels
telem> Not ready to fly: Sensors not operational.
commander: INVAL: ARMING_STATE_STANDBY_ERROR - ARMING_STATE_STANDBY
[px4io] default PWM output device
[i] PX4IO started
sensors_task: valid configs: 2 gyros, 1 mags, 2 accels
[i] Set PX4IO update rate limit: 400 Hz
mavlink_if0: mode: 0, data rate: 1000 B/s on /dev/ttyS1 @ 57600B
ver hwcmp match: PX4FMU_V2
[init] Vehicle type: VTOL
+ MAV_TYPE: unchanged
[px4io] mixer sent
[i] Mixer: /etc/mixers/firefly6.main.mix on /dev/pwm_output0
<fmuservo> MODE_6PWM
<fmuservo> set_pwm_rate 0 50 50
[fmuservo] starting
<fmuservo> adjusted actuator update interval to 20ms
[i] Mixer: /etc/mixers/firefly6.aux.mix on /dev/pwm_output1
<fmuservo> set_pwm_rate 0 50 50
<fmuservo> set_pwm_rate f 50 50
nsh: test: syntax error
fmuservo: subscribe to actuator_controls_1
fmuservo: subscribe to actuator_controls_3
<fmuservo> adjusted actuator update interval to 20ms
vtol_att_control: started
fw_att_control: Initializing..

[i] No addons script: /fs/microsd/etc/extras.txt

NuttShell (NSH)
nsh> navigator: Could not clear geofence
navigator: offboard mission updated: dataman_id=0, count=0, current_seq=0
sdlog2: log buffer size: 22528 bytes
gps: ubx msg 0x0103 len 16 unexpected
gps: ubx disabling msg 0x0103
gps: ubx msg 0x0601 NAK
gps: not using NAV-PVT
gps: VER hash 0xe0cc2322
gps: VER hw  "  00040007"
gps: VER sw  "                  7.03 (45970)"
gps: VER ext "                   7.03 (45969)"
gps: module found: UBX
telem> settings autosaved
position_estimator_inav: baro offs: 366
Assertion failed at file:armv7-m/up_hardfault.c line: 184 task: Idle Task
sp:     20008b50
IRQ stack:
  base: 20008b9c
  size: 000005d8
20008b40: 080ae3be 20008b50 2000924c 08080545 080ae484 000000b8 0807a43d 00000010
20008b60: 200068d4 00000003 20009154 080807d1 080807bd 0808c273 e000ed24 08089f01
20008b80: 00000000 00000010 20008e90 000002a7 a7104354 08080741 20008aac 00000000
sp:     080884b1
User stack:
  base: 2000924c
  size: 000003e8
R0: 0808d6b9 00000011 00000000 080884b1 0808844f 000003e8 08088139 00000003
R8: 20008b1c 00000078 20008b60 000003e8 08088139 20008b14 0807c565 080ae3d9
xPSR: 20008b44 BASEPRI: 00000007 CONTROL: 00000000
EXC_RETURN: 080884b1
```

Note: This is a Pixhawk I just received (ETH) and haven't tried a second one yet so I can't rule out a board specific issue yet.

kd0aij, 2015-04-25:
perhaps this is related: https://groups.google.com/forum/#!msg/drones-discuss/LeRqYliTKwU/SvvIqSDGROgJ

davids5, 2015-04-25:
@SimonWilks I will commit the fix @kd0aij  mentioned please stand by....

SimonWilks, 2015-04-25:
I'm not home now and won't be able to test this till tomorrow morning. Will
let you know how it goes then.

davids5, 2015-04-25:
@SimonWilks  - The fix has been pushed. Thank you.

SimonWilks, 2015-04-26:
I just finished testing with the updated Nuttx and this fixes the issue. Thanks guys!

closed 2015-04-26 by SimonWilks