PX4/PX4-Autopilot #2093: 8th chanel on mix-file does freeze the whole sbus-output

severinleuenberger, 2015-04-26:
First at all - my configuration works very well and i am very happy with it. 

I use the actual PX4 master-dev-build from April 25th. 

There I recognized a very strange issue. 
All the sbus out chanels work great, but as soon as I adress the chanel 8, the whole servo-ouput freezes. 

Here the setup on my *.mix file, that does work fine: 

###

NG1 Rev2 Aerodynamics - Delta-wing and Rudder mixer for PX4FMU
=================================================
This file defines mixers suitable for controlling a delta wing aircraft with rudder using PX4FMU.

Elevon left left CH 1
---------------------
Differential roll more up, less pitch

M: 2
O:      10000  10000      0 -10000  10000
S: 0 0  -7500  -2500      0 -10000  10000
S: 0 1   4000   4000      0 -10000  10000

Elevon left middle CH 2
-----------------------
less roll, more pitch

M: 2
O:      10000  10000      0 -10000  10000
S: 0 0  -4000  -4000      0 -10000  10000
S: 0 1   8000   8000      0 -10000  10000

Elevon right right CH 3
-----------------------
Differential roll more up, less pitch

M: 2
O:      10000  10000      0 -10000  10000
S: 0 0  -2500  -7500      0 -10000  10000
S: 0 1  -4000  -4000      0 -10000  10000

Elevon right middle CH 4
------------------------
less roll, more pitch

M: 2
O:      10000  10000      0 -10000  10000
S: 0 0  -4000  -4000      0 -10000  10000
S: 0 1  -8000  -8000      0 -10000  10000

Rudder mixer CH 5
-----------------------
Thix mixer enables Rudder for manual yaw-steering  

M: 1
O:      10000  10000      0 -10000  10000
S: 0 2  10000  10000      0 -10000  10000

Airbrake CH 6 
-------------
M: 1
O:      10000  10000      0 -10000  10000
S: 0 4  10000  10000      0 -10000  10000

Motor speed mixer CH 7
----------------------
Channel group 0, channels 3 (thrust) is passed directly through to the output.

M: 1
O:      10000  10000      0 -10000  10000
S: 0 3      0  20000 -10000 -10000  10000

###

so far so good, 

As soon as I ad an 8th chanel, the whole sbus output is jamed, all servos freezes to their extreme positition

###

NG1 Rev2 Aerodynamics - Delta-wing and Rudder mixer for PX4FMU
=================================================
This file defines mixers suitable for controlling a delta wing aircraft with rudder using PX4FMU.

Elevon left left CH 1
---------------------
Differential roll more up, less pitch

M: 2
O:      10000  10000      0 -10000  10000
S: 0 0  -7500  -2500      0 -10000  10000
S: 0 1   4000   4000      0 -10000  10000

Elevon left middle CH 2
-----------------------
less roll, more pitch

M: 2
O:      10000  10000      0 -10000  10000
S: 0 0  -4000  -4000      0 -10000  10000
S: 0 1   8000   8000      0 -10000  10000

Elevon right right CH 3
-----------------------
Differential roll more up, less pitch

M: 2
O:      10000  10000      0 -10000  10000
S: 0 0  -2500  -7500      0 -10000  10000
S: 0 1  -4000  -4000      0 -10000  10000

Elevon right middle CH 4
------------------------
less roll, more pitch

M: 2
O:      10000  10000      0 -10000  10000
S: 0 0  -4000  -4000      0 -10000  10000
S: 0 1  -8000  -8000      0 -10000  10000

Rudder mixer CH 5
-----------------------
Thix mixer enables Rudder for manual yaw-steering  

M: 1
O:      10000  10000      0 -10000  10000
S: 0 2  10000  10000      0 -10000  10000

Airbrake CH 6 
-------------
M: 1
O:      10000  10000      0 -10000  10000
S: 0 4  10000  10000      0 -10000  10000

Motor speed mixer CH 7
----------------------
Channel group 0, channels 3 (thrust) is passed directly through to the output.

M: 1
O:      10000  10000      0 -10000  10000
S: 0 3      0  20000 -10000 -10000  10000

Parachute CH 8 
--------------

M: 1
O:      10000  10000      0 -10000  10000
S: 0 2  10000  10000      0 -10000  10000

###

And it doesnt help wheter I change the source of chanel 8 (I tried everything - to take RC aux1 from control group 3, or Flaps from Control Group 0 or what ever), I tried also without connected servos on sbus chanel 8. No luck. 

So there I am with my analysis and think, this must be a bug. 

Best wishes
Severin

severinleuenberger, 2015-04-28:
can anyone please doublecheck, if the sbus servo-output freezes with the second *.mix-file-content I pasted below. 

I think this is a real issue/bug there. I double-checked at least 7 times myself. 

By now, i can only use 7 chanels instead of 16. 

(just as reminder - to get sbus running you have to add under sd-card etc\mixers.txt just the line
px4io sbus1_out)

Maybe this bug existed since the chan16 branch merged. I didnt tested this one before (sorry). 

Best, Severin

AndreasAntener, 2015-04-29:
When exactly does your output freeze? As soon as it's booted? Or after you arm?

severinleuenberger, 2015-04-29:
Booting is normal, it is just the sbus out who freezes after arming. And its not px4 who freezes totally. I can disarm, and I can see all the values trough QGC. It is really just the px4 signal out on sbus (maybe on pwm as well, but i didnt tested this).

AndreasAntener, 2015-04-29:
Ok, need to get my sbus stuff that's lying around somewhere.. Then I can probably do a test today

AndreasAntener, 2015-04-30:
It's broken, also for the regular outputs. IO suddenly sets 0 for every output.
This seems to be the commit that broke it: https://github.com/PX4/Firmware/commit/f8c8876642857520e85ec8ca027146c11b4301e3

severinleuenberger, 2015-05-01:
@AndreasAntener Thanks for double checking and analyzing
@LorenzMeier  could you flag this as a bug for all plattforms?

AndreasAntener, 2015-05-05:
@severinleuenberger we had to revert the commit for the sbus 16 chan update to make this work again.. At least 8 output mixer files should now work again. Closing this too.

closed 2015-05-05 by AndreasAntener