PX4/PX4-Autopilot #2099: Implement battery failsafe

tumbili, 2015-04-27:
I it would be nice for the user to specify a battery charge threshold at which the plane enters RTL mode.

tumbili, 2015-04-27:
@AndreasAntener @thomasgubler FYI

severinleuenberger, 2015-04-27:
Please consider, that RTL is not always the best and most secure method for operations. 
We would appreciate if you leave a door open to not use this function (perhaps use 0.0v for RTL - means desabling RTL). Thanks.

tumbili, 2015-04-27:
@severinleuenberger Sure, this will be optional and probably disabled by default.

AndreasAntener, 2015-04-28:
About the OBC failsafe, regarding the discussion yesterday:
https://pixhawk.org/users/failsafe

I'm not saying that this has to be exposed to the user in QGC, I think it's good to limit that to the very basic options. But it makes sense to consider all failsafe options in the firmware implementation so the whole state handling is in one place.

(Thomas is better suited to point at the code if necessary :))

thomasgubler, 2015-04-28:
The 2 most important pointers to the code:
  1. Here the detected failures are evaluated and a mode is requested from the navigator (search the individual variables e.g. gps_failure to see how they are set https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L1637)  https://github.com/PX4/Firmware/blob/master/src/modules/commander/state_machine_helper.cpp#L529
  2. here the mode set above is evaluated https://github.com/PX4/Firmware/blob/master/src/modules/navigator/navigator_main.cpp#L425 you can see that already some parameters are evaluated as well (for switching between normal and OBC mode)

The first part is more concerned about which state we *can* go in given the sensors/subsystems which still are healthy. The 2nd one is about *what* we do exactly in this state

tumbili, 2015-04-28:
@thomasgubler Thanks for the pointers, I will look into it and probably get back to you.

LorenzMeier, 2016-01-06:
@tumbili This is something you should really, really look into next week.

LorenzMeier, 2016-04-23:
Done!

closed 2016-04-23 by LorenzMeier