PX4/PX4-Autopilot #2105: POSCTL is broken in latest master

kd0aij, 2015-04-29:
I suspect that magnetometer calibration is broken in latest master/QGC.
Flashing continuous build binary 201504030932 and recalibrating the mag using an old version of QGC corrected the POSCTL problem.

DonLakeFlyer, 2015-04-30:
Can yor check before and after mag cal parameter values to see if they differ significantly?

kd0aij, 2015-04-30:
after:
x + CAL_MAG0_ZSCALE: 1.0000
x + CAL_MAG0_YSCALE: 1.0000
x + CAL_MAG0_XSCALE: 1.0000
x + CAL_MAG0_ZOFF: -0.4443
x + CAL_MAG0_YOFF: 0.1024
x + CAL_MAG0_XOFF: -0.0517
x + CAL_MAG0_ROT: -1
before:
1    1    CAL_MAG0_ROT    -1    6
1    1    CAL_MAG0_XOFF                    -0.125694    9
1    1    CAL_MAG0_XSCALE                      1.13073    9
1    1    CAL_MAG0_YOFF                   -0.0950166    9
1    1    CAL_MAG0_YSCALE                      1.06214    9
1    1    CAL_MAG0_ZOFF                    0.0416413    9
1    1    CAL_MAG0_ZSCALE                      1.13307    9

DonLakeFlyer, 2015-04-30:
Wait a minute. You kept the firmware the same and were just using different version of QGC? That won't make any difference to the values. QGC just puts a visual on top of calibration. It doesn't actually do any of the calibration itself or write any of the values. It just starts it and watches the log messages go by. The calibration all happens on the firmware side. So your fix more likely came from the firmware update not going back to old QGC. You should be able to go to a new QGC with the new firmware and it will work fine.

AndreasAntener, 2015-04-30:
As I understood it, Mark flashed an old firmware from 201504030932. I guess there's no problem in QGC. But the thing is, with an older firmware the calibration routines from QGC can be started but you won't get the right feedback.

DonLakeFlyer, 2015-04-30:
> with an older firmware the calibration routines from QGC can be started but you won't get the right feedback.

QGC should fall back to text based calibration in this case. Is it not doing that? It's still trying to show visual calibration with the pictures?

DonLakeFlyer, 2015-04-30:
> Mark flashed an old firmware from 201504030932.

Ah, that was an old version. I saw "continuous" and thought new. I though this Issue was reported on the QGC github. Sorry, I bounce back and forth so much I get confused as to what I'm doing sometimes. Let me look again.

AndreasAntener, 2015-04-30:
@kd0aij we did a calibration today on the Firefly and mag worked.
```1	50	CAL_MAG0_ID	73225	6
1	50	CAL_MAG0_ROT	0	6
1	50	CAL_MAG0_XOFF	                 0.253498	9
1	50	CAL_MAG0_XSCALE	                  1.04548	9
1	50	CAL_MAG0_YOFF	               -0.0811536	9
1	50	CAL_MAG0_YSCALE	                 0.930561	9
1	50	CAL_MAG0_ZOFF	               -0.0870424	9
1	50	CAL_MAG0_ZSCALE	                 0.992626	9
1	50	CAL_MAG1_ID	131594	6
1	50	CAL_MAG1_ROT	-1	6
1	50	CAL_MAG1_XOFF	                 0.089917	9
1	50	CAL_MAG1_XSCALE	                        1	9
1	50	CAL_MAG1_YOFF	                -0.323147	9
1	50	CAL_MAG1_YSCALE	                        1	9
1	50	CAL_MAG1_ZOFF	                 0.716464	9
1	50	CAL_MAG1_ZSCALE	                        1	9
1	50	CAL_MAG2_ID	0	6
1	50	CAL_MAG2_ROT	-1	6
```

@DonLakeFlyer : ok, I didn't know that QGC should be backwards compatible there, to be honest I haven't tried it.

DonLakeFlyer, 2015-04-30:
> we did a calibration today on the Firefly and mag worked

Using latest master firmware?

DonLakeFlyer, 2015-04-30:
> QGC should fall back to text based calibration in this case. Is it not doing that? It's still trying to show visual calibration with the pictures?

Only the newest versions of QGC have this ability. Went in a few weeks back.

AndreasAntener, 2015-04-30:
Pretty much latest: https://github.com/PX4/Firmware/commit/bd1c3363df44170523c68fd87ee19f2582a5e8fc
And also daily QGC builds

DonLakeFlyer, 2015-04-30:
So no longer a problem. Close this?

AndreasAntener, 2015-04-30:
I'm not sure. We have very flaky results ATM. There might still be something wrong with the calibration on the firmware side. I can only say that it worked today.

kd0aij, 2015-04-30:
my observation was just that POSCTL no longer worked correctly after flashing latest master and doing a complete calibration using latest QGC. Based on Lucas' post here: https://groups.google.com/d/msg/px4users/jAY1PwR_u6A/l4p3F9v_i0UJ
I decided to try to replicate his results, and the version he specified worked properly for me too. I reverted to an old version of QGC to recal the mag because it hung up when I tried to use latest QGC. I expect Lucas just reloaded all of his old parameters from a file. I was unable to do that since my FMUv1 was just moved to a different airframe.

kd0aij, 2015-04-30:
I also encountered bad POSCTL behavior in a Pixhawk with latest master. The same procedure fixed the problem with FMUv2 also.

DonLakeFlyer, 2015-04-30:
> because it hung up when I tried to use latest QGC

Details? Hung up means what?

> POSCTL no longer worked correctly after flashing latest master and doing a complete calibration using latest QGC

Did you try it again with todays firmware?

kd0aij, 2015-04-30:
Hung up meant that QGC was still saying "rotate for 20 seconds" several minutes after starting the 3rd (I think) orientation.
Haven't tried anything new, since I haven't seen any reports that the problem has been diagnosed or a fix has been made. Did I miss something?

DonLakeFlyer, 2015-04-30:
Got it thanks. New QGC should have switched to text mode cal when it detected older firmware. It didn't so it got confused. That's a bug.

@AndreasAntener Said the re-cal'ed with todays firmware and it worked.

AndreasAntener, 2015-04-30:
Ah, this issue is of course also about POSCTL, we didn't try that.

kd0aij, 2015-04-30:
I should point out that the heading looked correct on the bench in QGC even when POSCTL was "flying away" during flight tests. As Lucas was asking: Have there been any changes to the firmware in the past month which... may have broken POSCTL? The reason I suspected the magnetometer calibration was because it has changed recently and bad heading info can certainly screw up POSCTL.

If anyone can suggest particular commits which might be relevant, it might not be necessary to do a bisection to find the culprit.

DonLakeFlyer, 2015-04-30:
Any logs to pull heading data out of to see what happened?

kd0aij, 2015-05-01:
I've gotten confused with all the permutations, and it looks like I didn't save the logs from the problem flights because I assumed I had broken something myself. 

I'll reflash with latest master, recalibrate and see if I can reproduce the POSCTL problem.

kd0aij, 2015-05-01:
Latest master: Firmware Version	2a46e0f0b6a9521015e05b87209de7f9604b9205
with "old" calibration parameters.
Logfile: https://copy.com/AgbBtmHaiXcpBGoa (log muncher not accepting uploads atm)

Toilet-bowling develops after 10-15 seconds in POSCTL mode.

This seems to be caused by thrust-related EMI affecting the Pixhawk internal magnetometer.

But that was not a problem with FMUv1 installed in same location and older version of code (as shown in the plots below, the EMI influence on the mag was similar with FMUv1).

[image]

[image]

[image]

[image]

This is starting to look like more of a tuning problem than a bug (remember I swapped my FMUv1 with a Pixhawk and had to copy parameters from one config to the other), but what could have caused Lucas to need to retune POSCTL? As far as I know, he changed nothing except to upgrade his firmware.

SimonWilks, 2015-05-01:
Did you check in your spam folder for the Email? I just tried it with your log and it worked. Can't access the logmuncher log file from where I am at the moment so I can't check if it actually had an issue.

Plots: http://dash.oznet.ch/view/rkGhRojaLS5KagCoh6cUTA

SimonWilks, 2015-05-01:
Have you ruled out vibrations? They aren't massive but not small either.

kd0aij, 2015-05-01:
This is an older FMUv1 log on the f450 frame which shows no EMI problems with the internal magnetometer: http://dash.oznet.ch/view/Abu5WFQQTcTTC3ocXVLHL7#Thrust_PLOT
Why would newer logs show a problem with the same airframe? I don't see how minor calibration differences would affect this, since it's the magnitude of the mag vector which is affected. Could this be a problem I introduced when swapping the v1 for the Pixhawk?

kd0aij, 2015-05-01:
Connected external mag and recalibrated compass with latest QGC and master 2a46e0f
No coupling from thrust to abs mag now.

Calibration was quicker (now shorter than 20 seconds per orientation?) but the visuals still display the AP orientation instead of vehicle pose.  I also couldn't find a way to specify the external mag orientation; is this just not exposed by QGC? And I can't tell from QGC what a rotation value of "-1" means.

Magnetometer appears to be working correctly on the bench, but I couldn't tell from QGC or the console whether one or both was being used. (INAV_enabled=1).  Raining here, flight/POSCTL test will have to wait till tomorrow.

@SimonWilks will fly the next tests with smaller props to see if vibration levels drop back down to those in the log mentioned above. I didn't check the balance on the larger props, so that's likely to be the cause of the higher levels.

kd0aij, 2015-05-01:
weather cleared up: POSCTL working fine in position hold and small low speed circle.
http://dash.oznet.ch/view/GMfJwebYE2PhR9tNnD5TnV

Root cause of POSCTL problem must have been the degraded performance of internal mag. Still no clues as to why the internal mag now shows so much EMI.

So for Firmware Version	2a46e0f0b6a9521015e05b87209de7f9604b9205
I am no longer observing POSCTL as broken, but the internal magnetometer is no longer adequate.

DonLakeFlyer, 2015-05-02:
Sorry everyone, super busy today. I’ll send an email tomorrow going over how all the parameters work for calibration (things like rotation and so forth).

tumbili, 2015-05-03:
@kd0aij POSCTL works fine for me with latest master and sensor calibration with latest QGroundControl. Tested outdoors with IRIS.

kd0aij, 2015-05-03:
@tumbili Is it possible something has changed which is affecting only the internal magnetometer? I'd like to understand why I can no longer use the internal compass on this quadcopter when logs showed no coupling between thrust and mag field in the past.

tumbili, 2015-05-03:
@kd0aij Hang on, I will test fly with internal

tumbili, 2015-05-03:
@kd0aij I can't confirm the problem with the internal mag. I disconnected the external one but there is not difference. What do you actually mean by "degraded performance"?

kd0aij, 2015-05-03:
"coupling between thrust and mag field"
Before I swapped the FMUv1 with the Pixhawk, POSCTL worked well on the internal mag.
After swapping, flashing latest master and recalibrating, the Pixhawk internal mag reacted strongly to changes in thrust. The difference must be physical; perhaps the small difference in magnetometer location and power module/battery leads was enough to cause the problem.

tumbili, 2015-05-03:
@kd0aij Could well be. Did you try to checkout to an old version of firmware and QGroundControl?

kd0aij, 2015-05-03:
I at first thought that POSCTL worked properly with the Pixhawk/F450 airframe on reverting to older firmware and original mag calibration. That is why I posted this issue (and what I state in my first comment above). Perhaps we'll hear more observations from Lucas soon (he was the first to suspect a problem with master),

kd0aij, 2015-05-04:
Just verified that, with no parameter changes (calibrated using latest master/QGC) and no external magnetometer, the **pitch-yaw coupling is not significant** in Firmware Version 368a21d from the 2015-04-03 continuous build for both px4fmu-v1_default and px4fmu-v2_default. 
This is what the coupling looks like using latest master (very obvious in flight):
[image]
Log file: https://copy.com/uRQMYRZNBv0FRkA1
parent folder: https://copy.com/O2NqQDXuk99Tmi1N

tumbili, 2015-05-06:
@kd0aij Can you try to the same experiment using the new simple attitude estimator? From what you say it sounds like an attitude estimation problem.

kd0aij, 2015-05-06:
yes, just flashing the attitude_estimator_q branch fixed the yaw problem with no change to calibration. So that points to a regression in ekf_attitude_estimator.

LorenzMeier, 2015-05-06:
Correlation does not imply causation 8). The different formulations of the filters might be more susceptible to sensor calibration issues. Would you mind re-calibrating with current master (after Roman has fixed an issue yesterday)? Please re-calibrate gyro, accel and mag and let us know if that fixes it for you.

LorenzMeier, 2015-05-06:
The reason I'm less sure about regression in the estimator is that I don't see any relevant changes there.

kd0aij, 2015-05-06:
Agreed on "correlation", which is what I meant by "points to". Your observation that nothing relevant has changed in the ekf att estimator "points to" the regression being somewhere else.

Certainly all of the problems I've been seeing recently could be caused by bad calibration, but between all of the changes in  pre-arm checks and calibration in both master and QGC, the behavior of the system has just seemed flaky to me. 
In terms of "user experience" the score is currently so low that I'm now loathe to even attempt recalibration. Here's hoping that the next time around, I'll have a less frustrating experience :)

LorenzMeier, 2015-05-06:
If you do, make sure to use a QGC daily build and not one of the stable versions.

kd0aij, 2015-05-06:
thanks, have been careful lately to do that

kd0aij, 2015-05-06:
:+1: used latest daily build of QGC to flash master on Pixhawk in F450 quad, and recalibrated gyro/accel/compass with external mag disconnected. Calibration process went smoothly, user experience good. Verified that the attitude_estimator_ekf was running by default.
Flight test was good, no noticeable yaw issues.
@tumbili Question: As Lorenz points out above, the yaw problem appears to have been the result of bad sensor calibration. What was broken before, and should that particular bug affect the EKF more than other estimators?

kd0aij, 2015-05-06:
@LorenzMeier If I reconnect the external compass, how can I go about calibrating it and validating flight performance with both compasses? This is a necessity if we allow in-flight failover to the internal compass. I'd rather not have to manually compare parameter values after I recalibrate, or manually substitute the values I just validated for the new ones.

LorenzMeier, 2015-05-06:
You need to recalibrate the system whenever you unplug or replug the external, and because the lack of enforcement has caused crashes in the past I don't want to change the approach there.
If you want to fly both compasses you would need to quickly re-calibrate after changing your system configuration. With the latest version of the assistant in QGC that should only take a minute.

If you really want to trick the system for testing purposes you can comment out the check here:
https://github.com/PX4/Firmware/blob/master/src/modules/commander/PreflightCheck.cpp#L94-L99

I don't want to add a default option though because of the referenced crashes this change is preventing now.

If you want to simulate a failover you need to disconnect while armed.

kd0aij, 2015-05-06:
@LorenzMeier Thanks, this is making better sense now. So the required calibration procedure (for me) really involves:
1) Calibrate with both mags connected
2) fly
3) disconnect external while still armed
4) fly again. 
(in the recent past, even when things looked good on the bench, the copter didn't fly well)

That's practical, if the external mag isn't neatly buttoned up inside the airframe. 

Would be much more user-friendly if there were a way to do that without messing with a DF13 connector. 

Perhaps I'm being paranoid and we can rely on the calibration process alone i.e. flight testing is not required at all. Personally, I'd like to know whether the internal magnetometer is good enough for RTL to work properly.

tumbili, 2015-05-06:
@kd0aij @LorenzMeier I'm not saying that the problem was the sensor calibration! You should calibrate your sensor with latest master and test both attitude estimators again. If you still have the yaw problem with one estimator then we need to investigate further. I don't think that there is a relation to the calibration fix which I have done yesterday. The calibration routine was broken completely which is why also the preflight checks triggered.

kd0aij, 2015-05-06:
@tumbili My last flight did just that for the EKF attitude estimator (as I stated above). Given the yaw problem is no longer present with EKF and was never significant in older versions of master (or with the Q estimator), what would you think caused the problem?

LorenzMeier, 2015-05-06:
@kd0aij So if I interpret your comments correctly the system is behaving now? The reason I ask is that the EKF can be thrown off by a bad mag calibration in roll and pitch.

kd0aij, 2015-05-06:
I clarified my comment above: https://github.com/PX4/Firmware/issues/2105#issuecomment-99486995
Let me know if it's still unclear.

tumbili, 2015-05-06:
@kd0aij Ok, and do you happen to have the commit ID of the firmware you flashed before?
I just want to know which change it was that resolved this issue.

kd0aij, 2015-05-06:
From the log referenced here: https://github.com/PX4/Firmware/issues/2105#issuecomment-98887105
the commit is:
Firmware Version	0b5e6bdf9755cf530bf425b0b77db2c5374a279c

May I suggest that each commit include detailed comments stating whether it adds a feature, or fixes a bug, and specifically what the feature or bug is? Then other devs might be able to help out more with finding regressions.

LorenzMeier, 2015-05-12:
I'm closing this for now since we couldn't repro in quite a number of flights. I'm happy to assist in tuning the airframe if required. I also think we should switch to the q estimator plus INAV on fmuv1

closed 2015-05-12 by LorenzMeier