PX4/PX4-Autopilot #2107: Logging doesn't work

DrTon, 2015-04-30:
Logging doesn't work in current master (tested on multirotor config). I suspect because of 6d41b5d. If this is true then "stable" is also affected. @LorenzMeier, maybe it's time to go away from descriptor-based topics and do more careful testing of "stable"?

tumbili, 2015-05-03:
@DrTon Logging works for me on latest master (Tested with IRIS). Can you retry your setup? And what exactly do you mean by "not working"?

DrTon, 2015-05-03:
Not working means no log files after flight. Have you recompiled NuttX and using default parameters (INAV_ENABLED=1)? Will try again, maybe something already changed.

tumbili, 2015-05-04:
@DrTon Recompiled NuttX and made sure INAV is running. Still logging works. Let me know if it works for you.

DrTon, 2015-05-11:
I think I know why it's "not working". Right now (11 may 2015) I got the following:
```
telem> [sdlog2] log dir: /fs/microsd/log/2015-04-23
telem> [sdlog2] starting: 13_05_52.px4log
```
IMO it's better to have sess001/log001 in case of no GPS...

kd0aij, 2015-05-11:
#2084 may be related

LorenzMeier, 2015-05-17:
I've addressed this in https://github.com/PX4/Firmware/pull/2191. There is fundamentally nothing intelligent we can do about clock drift, other than offering developers the choice to rely on GPS only. For outdoor operations the RTC backed mode makes a lot of sense though.

LorenzMeier, 2015-05-17:
Closing, please refer to the PR.

closed 2015-05-17 by LorenzMeier

DrTon, 2015-05-17:
One day error after day of sleeping is not a DRIFT :) it's a bug somewhere in RTC.

kd0aij, 2015-05-17:
One would hope that the battery backup would last for more than a day or
two...
The MS5621 appears to have a 5.5mah capacity, while the processor has a
worst-case draw of 11ua.
5.5mah/11ua = 20.8 days
Unless, of course, something else is loading Vbat...

kd0aij, 2015-05-17:
BTW, typical Vbat mode current draw is spec'd at 1.6ua, for an endurance of
143 days

LorenzMeier, 2015-05-17:
@DrTon I can't reproduce - what steps do I need to do to recreate your case?

kd0aij, 2015-05-17:
Was this perhaps an FMUv1?