PX4/PX4-Autopilot #2111: sometimes first boot ends with 5 beeps before freeze, second boot works fine

severinleuenberger, 2015-04-30:
From time to time my board freezes on booting. It ends with 5 beep tones. 
After restarting the board, everything works normally. I couldnt figure out, what was the cause. 
As I heard, I am not the only one with this issue. 

I use the latest dev-build from Apr 29 with a dropix/pixhawk-board.

SimonWilks, 2015-04-30:
Is it a constant tone or does it vary? Is it something you think you could trigger on the bench after a bunch of reboots? If yes connecting a debug cable to see the console output might give a hint as to what's failing.

severinleuenberger, 2015-05-01:
First the normal startup melody is played, but then after a second more, the error-message tone comes. It is a constant tone with 4 or 5 beeps. 
It alwasy comes only on the first boot. When disconnecting the battery and reconnecting again, it doesnt beep anymore. 
What could be an issue: I recognized trought QGC, that Batery Voltage increases slowly (in 3-4 seconds) from 17v until 25v (6S Setup). Maybe this has something to do with it? Because on PX4 I declare a 6S Setup, and 17v wold be an empty lipo-battery.

sjwilks, 2015-05-01:
Yes it does sound like battery then. I'll take a closer look when I'm in
the office. We might need a slight delay before doing the voltage check.

The board shouldn't freeze though. By freeze you mean no connection via QGC
or you can't arm?

severinleuenberger, 2015-05-01:
I just coudnt arm i think. Yes maybe with a delay  of 4 seconds for the battery voltage  check it could be solved.

SimonWilks, 2015-05-01:
I had a closer look and the 5 low beep tones signal an error during startup when the rcS script runs, the battery voltage tune is very different (fast continuous beeps that never stop). Unfortunately this error tone is used for several different failure cases, such as FMU, IO, IO firmware update, uavcan, mkblctrl etc.

I think we need to increase what's written to the bootlog.txt file.

SimonWilks, 2015-05-01:
This should at least help identify which component is failing: #2114

severinleuenberger, 2015-05-01:
very cool! I will try this as soon as possible. Hopefully get the console merge working this time ;-)

severinleuenberger, 2015-05-04:
I have now the analysis from Serial5 debug via FTDI to TeraTerm. 
So most startups look like this: 
[image]

But every third or fourth startup brings ghe following error: 
[image]

So it seems to be the SD-Card somehow. 
mixer: error loading mixers from /fs/microsd/etc/mixers/rev2.mix: Operation not permitted
[i] Error loading mixer: /fs/microsd/etc/mixers/rev2.mix

What i do not understand, is why it doesnt happen every time. Any help?

SimonWilks, 2015-05-04:
I experienced pretty much the same thing today. Every second or third boot I'd get this error too. One time (could have been more, I didn't notice) I was getting an SD card  mount failure message. We had a second board also complain about an SD issue with the same mounting error (out of memory).

Will try and do some more tests tomorrow but something looks fishy. I was using latest master.

LorenzMeier, 2015-05-04:
Could you both please try what happens if you disconnect the FTDI cable (and any other serial port back powering) and completely unplug and re-plug the USB connector? It could be a card-level brownout.

SimonWilks, 2015-05-05:
@LorenzMeier seeing it would have been the second bad SD card within a couple of hours on two different autopilots I didn't want to believe it was coincidental, but it seems it was. Every second boot gave me mounting issues even when only powered by USB. A new SD card resolved the issue.

The mixer error is odd though. Every second boot this morning was failing and then when I returned later on (just plugged it in again, no changes) it worked and I haven't had it happen since. After that I addressed and fixed the microSD.

SimonWilks, 2015-05-05:
Out of three Pixhawks all three have exhibited the mixer issue. One was then stopped failing, another was always failing regularly and still does and another was fine all morning and now just started doing it too.

Probably the next obvious question is what triggers the error message. Could there be some sort of delicate timing issue here?

LorenzMeier, 2015-05-05:
did you test with the console cable *disconnected*?

SimonWilks, 2015-05-05:
During microSD testing yes, disconnected. Will try the same with the board showing issues now. Ada suspects this might be related to #2113 and will update there shortly.

LorenzMeier, 2015-05-05:
If you revert f8c8876642857520e85ec8ca027146c11b4301e3 (```git revert f8c8876642857520e85ec8ca027146c11b4301e3```) does it work after? If yes, we should put this on master. I'm afraid I currently don't have time to do 8+ outputs on S.Bus, someone else will need to do this properly then.

closed 2015-05-05 by LorenzMeier