PX4/PX4-Autopilot #2113: IO doesn't load if mixer file has 8 outputs

AndreasAntener, 2015-05-01:
Maybe related to https://github.com/PX4/Firmware/issues/2093

IO fails if the mixer has 8 outputs assigned, this happens on start:
```
[px4io] mixer send error -1
mixer: error loading mixers from /etc/mixers/AERT.main.mix: Operation not permitted
[i] Error loading mixer: /etc/mixers/AERT.main.mix
```

AndreasAntener, 2015-05-05:
I have to add that although it seems to be calibrated, I get these messages now:
```
[07:41:38.102 - COMP:50] Critical: PREFLIGHT FAIL: ACCEL 0 SELFTEST FAILED
[07:41:38.166 - COMP:50] Critical: PREFLIGHT FAIL: ACCEL 1 SELFTEST FAILED
```

AndreasAntener, 2015-05-05:
@LorenzMeier I'm pretty sure that this commit broke it: https://github.com/PX4/Firmware/commit/f8c8876642857520e85ec8ca027146c11b4301e3
It works before that.

LorenzMeier, 2015-05-05:
@AndreasAntener Could you just revert the commit on master and verify that fixes it?

AndreasAntener, 2015-05-05:
I just did that, it fixed the issue at startup. But I would like to arm and verify correct operation, but having trouble getting through preflight checks now ;)

AndreasAntener, 2015-05-05:
Ok, 8 channels working: #2130

AndreasAntener, 2015-05-05:
Fixed by #2130

closed 2015-05-05 by AndreasAntener

severinleuenberger, 2015-05-06:
double-checked - the 8ch *.mix does work again! 
Thanks