PX4/PX4-Autopilot #2115: PX4 RC input/output: Stuttering issues

philipoe, 2015-05-02:
Hi all,

*Note: This is a discussion thread to clarify who else has seen this issue and to check who could/how it could be solved*

we operate larger-scale platforms (fixed wing with ~5m wingspan) that use Pixhawk, and we have for quite some time now (acutally with firmwares dating >1 year back) observed servo stuttering when using the Pixhawk outputs. We are quite certain that all platforms are affected by the issue, but it is not visible on many small platforms because of their small control surfaces (we could not visually see the issue on a 2.6m and 1.8m elapor test-rigs with the same configuration as the big plane, probably due to the low control surface inertia).

### TEST SETUP:
 - 1x PX4 FMUv1.7 and PX4IO v1.3 with Spektrum satellite receiver. Firmware stable from Nov 17th 2014 (97a1410ec99e880207e4ee6d2a03451c2e11f4cf) with custom modifications, but same issues observed with newer stable firmwares.
 - 1x Spektrum Standard RC receiver (AR8000).
 - Both connected to a central multiplexer, which also powers all servos (i.e. only signal is coming from PX4)
 - Oscilloscope connected to both PX4IO PWM output and Spektrum PWM output channel 2 (Aileron). Note that PX4IO and Spektrum receiver may output at different frequencies, therefore the signal may be shifted. Also note that in the first tests below the mixer applies a differential to the aileron channel on PX4IO, this is why the signals may look asymmetric.

### PRE-DISCUSSION:
We think that there are two seperate issues:
 - Issue 1 concerns purely the RC input/output side (Test 1, see below)
 - Issue 2 concerns the IO<->FMU communication (Test 2, see below) and the issues this brings into correctly communicating the RC inputs and outputs. This issue is the root cause of the servo stuttering that we see.

### TEST1: 
 - RC input sweeps are given via standard remote (Spektrum DX8)
 - PX4IO output rate is 50 Hz, Spektrum receiver output rate is 45Hz (therefore the signal shifts over time!)
 - PX4IO/FMU are in MANUAL mode, i.e. signal trace is PWM In -> PX4IO -> PWM Out.

PX4IO output rate 50Hz:
[image]

PX4IO output rate 100Hz:
[image]

PX4IO output rate 40Hz:
[image]

 - Observation/Explanation: Output signal has small dents/jumps at 50Hz output frequency. This worsens at 100Hz output frequency. It is obvious why this happens: The Spektrum remotes send their updates at 45Hz, which means that every once-in-a-while, the output will "update too fast" and there will not be a new input update from the spektrum remote. This is why the discontinuities exist. At 40Hz, the issue is completely gone.
 - **Question: Is it advisable (at least for MANUAL mode) to update the PX4IO outputs faster than we get input signals? I could imagine that this is not a significant issue, because the servos may often just do position control. Or should we in general limit the output frequency below the input frequency (say 40-42Hz)?**

### TEST2: 
We kept the PX4IO output frequency at 40Hz. The signal looks clean now (see Test 1 above), and there is NO servo stuttering when operating in MANUAL mode. However, when switching to the assisted modes (i.e. signal path is now PWM Input -> IO -> FMU -> IO -> PWM Output), the stuttering is still clearly visible. Note that we use a custom control-implementation on PX4FMU that runs at 100Hz, and that provides a simple assisted mode that will basically just feed-through the RC commands when the aircraft is at rest - it's a stability augmented system controller.

#### Test2a:
When in a non-MANUAL mode, the stuttering still exists when giving sweeps as RC inputs. The output signals show jumps (we did not record the signal however), and one can visually observe the stuttering on the aileron. 

#### Test2b:
To exclude the RC input as the source of the bug, we implement a small linear ramp on the outputs (that just goes from PWM 1150->1920us and then back to 1150) into our controller on PX4FMU, i.e. we are directly writing the actuators (actuator_controls_s). The signal trace is therefore FMU -> IO -> PWM Outputs. The resulting graph is seen below (the Spektrum receiver of course receives nothing during that test). The output looks rather clean, i.e. there are no jumps in the signal and the stuttering is also not visible at the aileron control surface.
[image]

#### Test 2c:
We programmed a similar linear ramp directly into the PX4IO RC signal decoding source, i.e. we replaced https://github.com/PX4/Firmware/blob/master/src/modules/px4iofirmware/dsm.c#L390 by the linear ramp. The signal trace is therefore PX4IO -> FMU -> IO -> PWM Outputs. The stuttering is clearly visible on the aileron, and the signal shows discontinuities.
[image]

### How to solve this: Support needed!
 - Given the tests above, we assume a communication issue in the PX4IO -> FMU trace/part that the RC signal has to go through. This is why we don't see the issue when in MANUAL mode and also not when writing the RC outputs directly on FMU.
 - Has anyone seen similar issues (as I said, most probably not visible on smaller platforms, we also do not visually "see" the issues on our 2.6m elapor test-rig).
 - We're not experts in the intra-board communication. **Can somebody with more knowledge on intra-board communication comment on what might cause this, i.e. how the commnication w.r.t. triggering the RC-signal communication works, and whether e.g. timings may be wrong there?**

Best,
Philipp

LorenzMeier, 2015-05-06:
Thanks for the really, really good test data. This kind of analysis is what really contributes to improving the system and having a fresh look at things.

What you are seeing in the data is the "notching" / aliasing + latency of the RC input all the way to the controller output. Because the plane controller only runs at 50 Hz its visible in the logs. I would guess that running the controller at 200 Hz is pretty much all it takes to get rid of this.

Commenting out this line should do the trick:
https://github.com/PX4/Firmware/blob/master/src/modules/fw_att_control/fw_att_control_main.cpp#L643

You shouldn't need to drive the servo at 100 Hz to get rid of of it, but let us know if that assumption is correct.

LorenzMeier, 2015-06-03:
Did you end up increasing the output rate?

closed 2015-07-03 by LorenzMeier