PX4/PX4-Autopilot #2119: unconstrained actuators.output may cause output suddenly inverse

taogashi, 2015-05-03:
I found that the rate_controller's output is unconstrained, mc_att_control_main.cpp, line 861:
```
_actuators.control[0] = (isfinite(_att_control(0))) ? _att_control(0) : 0.0f;
_actuators.control[1] = (isfinite(_att_control(1))) ? _att_control(1) : 0.0f;
_actuators.control[2] = (isfinite(_att_control(2))) ? _att_control(2) : 0.0f;
```

Later in px4io.cpp line 1174, we convert a float to int16_t:
```
for (unsigned i = 0; i < _max_controls; i++) {                                                                                                                       
        regs[i] = FLOAT_TO_REG(controls.control[i]);
}
```
and
```
#define FLOAT_TO_REG(_float)	SIGNED_TO_REG((int16_t)((_float) * 10000.0f))
```
In some extreme condition, _actuators.control will go above 3.2768, resulting an int16_t's overflow after multiply by 10000.0.
Such condition appears when a user like me to set a fairly large pid parameter, drone crashed...

tumbili, 2015-05-03:
@taogashi Thanks for reporting. Makes sense to me. @LorenzMeier Do you agree?

kd0aij, 2015-05-03:
this looks like the root cause of the instability problems in POSCTL: high rate demand leads to overflow

tumbili, 2015-05-03:
@taogashi Could you share a log of your crash?

LorenzMeier, 2015-05-04:
Good find. @tumbili I think we should cap the value from -1 to 1 before calling FLOAT_TO_REG() on it: https://github.com/PX4/Firmware/blob/master/src/drivers/px4io/px4io.cpp#L1175

commit e09f5d28 (2015-05-04): IO driver: Ensure comms protocol cannot get into integer overflow on bad control outputs. Fixes #2119.
closed 2015-05-04 by LorenzMeier