PX4/PX4-Autopilot #2122: Gimbal attitude control in fixed-wing mode not useable

AndreasAntener, 2015-05-04:
We're using actuator output group 2 in the FW attitude controller: https://github.com/PX4/Firmware/blob/master/src/modules/fw_att_control/fw_att_control_main.cpp#L1094
That and the gimbal app fight each other..

AndreasAntener, 2015-05-04:
This could to be a little tricky to fix because group 2 can have multiple sources who care about different things (e.g. gimbal and parachute). But we cannot separate them since it's the same topic.

LorenzMeier, 2015-05-04:
Sounds like we need to maybe give the gimbal its own control group?

AndreasAntener, 2015-05-04:
I only see this option ATM. Might also make sense from a safety perspective, e.g. you don't want the gimbal controller mess with your chute ;)
A "payload" group will always ask for this conflict.. I was wondering why there is a chute release part in the fw att controller. Where does it come from?

AndreasAntener, 2015-09-02:
@LorenzMeier 
Moving the discussion from Gitter here, the basic proposal is the following:
More payload groups:
- airframe configuration (flaps, gear, parachute, etc.)
- gimbal
- camera
- additional payload group 1
- additional payload group 2

I split gimbal and camera because I feel these are separate concerns and separate external APIs. Although I lack insight about what's there already in this area. E.g. how is the recently added camera trigger driver used?

Like you proposed, we would start with adding just one group for the gimbal. The existing payload group can then be used for airframe configuration.

Handling of the airframe configuration group needs to be centralized and the current handling (mostly in the FW controller) moved into this new app. For example: landing gear and parachute will controlled by various other parts from the system. How should we interface this?

LorenzMeier, 2015-09-02:
I'm still not convinced if camera and gimbal are two groups. The camera trigger is a GPIO and unrelated to actuator controls. Airframe seems trivial: You need a library (in a worker or running in the attitude controller, not as a real app) which evaluates high-level topics like some sort of failsafe topic and de-conflicts them based on airframe knowledge (e.g. it stops the propellers before shooting the parachute or it deploys and retracts the landing gear automatically based on laser measurements).

AndreasAntener, 2015-09-02:
I agree that we should just start with the use cases we have now.

I'm trying to understand how the airframe config would work within a worker, which sounds like the right approach: every controller app would start that worker, but it only runs once even if started from multiple locations. It will take inputs such as vehicle commands, auxiliary switches, vehicle states and control the config group accordingly.

AndreasAntener, 2016-05-02:
On 09.11.2015, at 15:53, Lorenz Meier <lm@qgroundcontrol.org> wrote:

> Hi
> 
> I guess that's a reminder we need to disambiguate between parachute and payload.
> 
> Anyone volunteering to introduce another actuator control group just for "fancy" airframe controls like parachutes? It should be group #4 and I would propose to implement it by default only in the PX4FMU driver and initially change the FMU mixer to output the parachute control on the 4th output. The AUX outputs would by default then see this use:
> 
> AUX1: gimbal roll
> AUX2: gimbal pitch
> AUX3: gimbal yaw
> AUX4: parachute
> AUX5: Lidar-lite or camera trigger
> AUX6: Lidar-lite or camera trigger
> 
> Does that make sense?
> 
> -Lorenz
> 

The airframe configuration groups could handle gear, parachute, direct servo control etc. The library (worker) will need to listen to commands and topics necessary to operate these.

LorenzMeier, 2016-10-19:
@AndreasAntener Starting to move there now with @julianoes

julianoes, 2017-04-18:
@bkueng FYI this is about actuator groups :)

closed 2018-02-22 by PX4BuildBot