PX4/PX4-Autopilot #2124: Failsafe no longer functioning

SimonWilks, 2015-05-04:
Failsafe (not RC failsafe) does not appear to be functioning. 

To test this I have taken a quad airframe (F330) and disabled the failsafe circuit breaker. On the SD card I created an /etc/extras.txt file containing:
```
pwm failsafe -c 1 -p 1300
pwm terminatefail on
```
To servo output 1 I have a servo attached which responds to throttle when armed. Triggering flight termination via the QGC widget shows the following on the debug console:
```
commander: forcing failsafe (termination)
commander: reset all non-flighttermination failsafe commands
```
however, the servo continues to function.

Looking into possible causes but if anyone is aware of a change that went in since January (we last tested this then) or a setup mistake above let me know.

SimonWilks, 2015-05-04:
Just flashed firmware from Jan. 2015 and it works. Time to hunt for the change.

SimonWilks, 2015-05-06:
I will complete testing tomorrow.

LorenzMeier, 2015-05-12:
This is fixed or not?

SimonWilks, 2015-05-12:
Yes, fixed.

closed 2015-05-12 by LorenzMeier