PX4/PX4-Autopilot #2125: Sensor calibration issues

kd0aij, 2015-05-04:
Internal mag calibration x/y/z scales are all 1.0

Are the IDs for MAG0, MAG1 handled correctly? if the external mag is disconnected after calibration it seems that MAG1 is then identified as MAG0

LorenzMeier, 2015-05-04:
That's correct, but it gets a different ID, which is used to look up the calibration parameters.

DonLakeFlyer, 2015-05-04:
> but it gets a different ID, which is used to look up the calibration parameters.

Only if you re-calibrate? This is disconnect without re-calibrating.

LorenzMeier, 2015-05-04:
different = unique, my bad. So you don't need to calibrate again.

kd0aij, 2015-05-05:
@LorenzMeier How do you suggest that one validate the calibration for both mags? If I disconnect the external mag, prearm checks fail, and I can't fly to test the internal mag. This testing may not be necessary for the normal user, but how else do we prove that the internal mag is working correctly?

LorenzMeier, 2015-05-05:
@kd0aij That's intentional. While the assignment of calibration parameters is done correctly based on the sensor's unique ID, the prearm check confirms wether what used to be MAG0 is still MAG0 and fails if its not.

The rationale is that you should not be flying a system that has changed configuration since you calibrated it. Either because you've added an external GPS and really should be using the external mag now, or worse, because the wiring to the external GPS might have come off.

As you showed yourself falling back to the internal mag is not quite ensuring a reasonable or even safe flight performance.

Presuming that this all works as designed, I'm closing this now. Please re-open if I missed something.

kd0aij, 2015-05-05:
Did you mean to imply that fallback to the internal mag is not allowed at any time, hence there is no need to test its calibration?

kd0aij, 2015-05-05:
Also, the internal magnetometer scaling question has not been addressed.

LorenzMeier, 2015-05-05:
@kd0aij Fallback to the internal mag is possible in-flight, but it will prevent arming. The scaling of 1.0 is ok / nominal for the LSM303D.

closed 2015-05-05 by LorenzMeier

kd0aij, 2015-05-05:
I conclude that flight testing with the internal mag is therefore not
supported.

Is the scaling intentionally forced to 1.0 for the LSM303D?

LorenzMeier, 2015-05-06:
No, that conclusion is not correct. Just calibrate the mag with the external mag disconnected and you will be allowed to use the internal one.
The scaling is not "forced" to 1.0, rather, the LSM303D does an automatic internal scale correction, so its scale is in fact always right / at 1.0.