PX4/PX4-Autopilot #2127: Accelcalibration reports failure but values are set

AndreasAntener, 2015-05-04:
With the master (#2098) I can do the accel calibration now but it reports a failure:
```
[21:04:25.736 - COMP:50] Critical: [cal] calibration failed: to apply calibration, se
[21:04:25.797 - COMP:50] Critical: [cal] calibration failed: accel
```

Not sure why. It seems to have done it and saved all the parameters for accel 0 and 1

tumbili, 2015-05-04:
@AndreasAntener Will also have a look at this tomorrow.

AndreasAntener, 2015-05-05:
I have to add that I get the following startup messages:
```
telem> PREFLIGHT FAIL: ACCEL #0 SELFTEST FAILED
telem> PREFLIGHT FAIL: ACCEL #1 SELFTEST FAILED
sensors_task: FAILED APPLYING accel CAL #0
sensors_task: FAILED APPLYING accel CAL #1

```

tumbili, 2015-05-05:
@AndreasAntener Looking at this now

tumbili, 2015-05-05:
@AndreasAntener I think the problem are the scaling values. They should be something around 1 but they are one order smaller in magnitude (0.1). The sensor drivers detects this and does not accept the scaling values.

tumbili, 2015-05-05:
Can you check your values?

AndreasAntener, 2015-05-05:
I don't quite understand what your saying, but my x/y scales are sure nowhere near 1 ;)

```
1	50	CAL_ACC0_ID	1246218	6
1	50	CAL_ACC0_XOFF	-0.109255	9
1	50	CAL_ACC0_XSCALE	0.0196716	9
1	50	CAL_ACC0_YOFF	0.191907	9
1	50	CAL_ACC0_YSCALE	-0.00694854	9
1	50	CAL_ACC0_ZOFF	0.803425	9
1	50	CAL_ACC0_ZSCALE	-0.99113	9
1	50	CAL_ACC1_ID	1114634	6
1	50	CAL_ACC1_XOFF	1.33859	9
1	50	CAL_ACC1_XSCALE	0.00669717	9
1	50	CAL_ACC1_YOFF	1.03434	9
1	50	CAL_ACC1_YSCALE	0.00297559	9
1	50	CAL_ACC1_ZOFF	-1.19388	9
1	50	CAL_ACC1_ZSCALE	-1.01322	9
```

tumbili, 2015-05-05:
@AndreasAntener I am slightly confused
math::Matrix<3, 3> accel_T_rotated = board_rotation_t * accel_T_mat * board_rotation;
This is the line that leads to the error.

tumbili, 2015-05-05:
Why exactly do you have to multiply from both sides?

tumbili, 2015-05-05:
https://github.com/PX4/Firmware/blob/master/src/modules/commander/accelerometer_calibration.cpp#L247

LorenzMeier, 2015-05-05:
offsets / scales are per sensor axis and completely agnostic of the board rotation. That part of the code looks incorrect to me, and all rotations should probably be stripped. The board rotation is only needed for guiding the user through the various positions, on a per-sensor basis its not relevant. Even more important, the sensor offsets should *not* depend on the board rotation param at all.

devbharat, 2015-05-05:
Isn't board rotation here the mounting orientation of IMU frame with respect to the body frame, and not the one guiding the user through the different body rotations to collect gravity in different axes ? 

^^Multiplications from two sides is actually derived in the comment at the beginning of the file - 
```
 * Calibrating using model:
 * accel_corr = accel_T_r * (rot * accel_raw - accel_offs_r)
 *
 * Actual correction:
 * accel_corr = rot * accel_T * (accel_raw - accel_offs)
 *
 * Known: accel_T_r, accel_offs_r, rot
 * Unknown: accel_T, accel_offs
``` 

rot here would represent the mounting of IMU to aircraft body frame.

LorenzMeier, 2015-05-05:
Fixed by #2131.

closed 2015-05-05 by LorenzMeier