PX4/PX4-Autopilot #2129: Commander 'system power' bug (?)

devbharat, 2015-05-05:
The system power check in commander.cpp
```
if (hrt_elapsed_time(&system_power.timestamp) < 200000) {
				if (system_power.servo_valid &&
				    !system_power.brick_valid &&
				    !system_power.usb_connected) {
					/* flying only on servo rail, this is unsafe */
					status.condition_power_input_valid = false;

				} else {
					status.condition_power_input_valid = true;
				}

				/* copy avionics voltage */
				status.avionics_power_rail_voltage = system_power.voltage5V_v;
				status.usb_connected = system_power.usb_connected;
			}
```

seems to work counter-intuitively. It gives ```NOT ARMING: Connect power module.``` error when power modules are actually connected, and not when they are disconnected.

When modifying the code to this:
```
if (hrt_elapsed_time(&system_power.timestamp) < 200000) {
				if (system_power.servo_valid &&
				    !system_power.brick_valid &&
				    !system_power.usb_connected) {
					/* flying only on servo rail, this is unsafe */
					status.condition_power_input_valid = true;

				} else {
					status.condition_power_input_valid = false;
				}

				/* copy avionics voltage */
				status.avionics_power_rail_voltage = system_power.voltage5V_v;
				status.usb_connected = system_power.usb_connected;
			}
```
(switching true and false), it works as expected. That is, status.condition_power_input_valid is true when battery and connected and USB is disconnected, and false otherwise.

The flags are actually filled in stm32 adc driver,
```
	// these are not ADC related, but it is convenient to
	// publish these to the same topic
	system_power.usb_connected = stm32_gpioread(GPIO_OTGFS_VBUS);

	// note that the valid pins are active low
	system_power.brick_valid   = !stm32_gpioread(GPIO_VDD_BRICK_VALID);
	system_power.servo_valid   = !stm32_gpioread(GPIO_VDD_SERVO_VALID);
```

but I did not check the pin values themselves, since the above fix seems to make it work.

tumbili, 2015-05-05:
@devbharat Are you

tumbili, 2015-05-05:
Sorry typing on the phone is not so easy :-) Are you sure there is nothing wrong with your setup?

tumbili, 2015-05-06:
From what I understand if you have the system powered over the power module you will have system_power.brick_valid = true and therefore the else statement is entered which sets status.condition_power_input_valid = true

devbharat, 2015-05-06:
Oh! this is to prevent the pixhawk from brownouts when you power just on the output rail! So you _need_ to use a power module in flight.

closed 2015-05-06 by devbharat