PX4/PX4-Autopilot #2132: Removing some modules prevents sucessful startup

crashmatt, 2015-05-05:
Removing tone_alarm completely prevents startup.  Removing some other modules prevents part of startup. I don't yet have a definitive list of what modules affect the startup in what way.  This seems to be an issue since the rcS script was re-ordered.

crashmatt, 2015-05-07:
tone_alarm seems too fundamental to avoid.  I tried putting an "if then fi" statement around all rcS instances of tone_alarm but that doesn't work properly.  There are more instances in other scripts.

some modules seem to define mavlink parameters that qgc demands to get started.  This somewhat defines a minimum set of functionality expected for something that is running PX4.  I don't yet know if these parameters are defined in the wrong place or if qgc needs modifying.

SimonWilks, 2015-05-07:
Why do you want to get rid of tone_alarm?

crashmatt, 2015-05-07:
@SimonWilks If you don't have any alarm connection on a board then why have a tone alarm?

This seems a bit picky since I can just output the alarm to an unused pin.  One of my goals is to improve the reliability of a simple PX4 configuration.  One way to help this is to remove everything that is not necessary.

For now this is low prioirty.  If I don't find anything more I will delete it.

LorenzMeier, 2015-05-07:
Since we want these hooks to be in I suggest you add a stub application which does nothing and returns OK. That's much more scalable than making the callout to the app optional - to me this is a fundamental architectural argument and not because I would oppose a general change.

Also check if the commander app is happy to not have a valid FD. If it isn't do the same there: instead of making the calls conditional, just return ok if the FD didn't exist.

crashmatt, 2015-05-07:
Thats a good plan.  Maybe creating a valid FD but disabling the driver operation is a way.

crashmatt, 2015-05-11:
Closing this issue for now.  It is mostly covered by #2152 and #2151.

closed 2015-05-11 by crashmatt