PX4/PX4-Autopilot #2140: Rate limiting has to be effective for rate and feedforward parts

LorenzMeier, 2015-05-07:
This log from @Zefz shows that the rate limiting is currently not effective for all gains:

http://dash.oznet.ch/view/VHo7DDhC9Sxy9FCEuQgJUK#A_RollRate_PLOT

This leads to an altitude loss with the new auto-scaling mixer. @tumbili This should be a quick fix.

tumbili, 2015-05-07:
@LorenzMeier @Zefz I don't think this is the problem. What about the number of satellites you had? Zero seems a bit too low for me.

tumbili, 2015-05-07:
Oh sorry, might have looked at the wrong thing

tumbili, 2015-05-07:
Yeah, sorry I didn't see the scaling of the axes

tumbili, 2015-05-07:
@LorenzMeier @Zefz I think you have a problem with the position controller. Why does it give you a constant velocity setpoint of -1? Actually the sign should be different.

Zefz, 2015-05-07:
@tumbili Can you try to fly the ```position_controller_inav``` with any of the attitude estimators and see if your Iris quad behaves properly?

I can run another longer test to collect more data.

tumbili, 2015-05-07:
@Zefz Yes, just flashed latest master, did calibrations and just heard my charger beeping to say that battery is full :-)

tumbili, 2015-05-07:
@Zefz Just flew with INAV enabled, no problems at all? Are you flying the latest master?

LorenzMeier, 2015-05-07:
Completely Independent of how the Position Controller operates rate limiting is not working, or it is applied after logging. This issue is about fixing this, so please focus the discussion / effort here on it. @Zefz flight performance is an entirely different topic and there are certainly usage problems involved.

tumbili, 2015-05-07:
@LorenzMeier I don't fully understand what is wrong with the rates. Where do you see any violation?

LorenzMeier, 2015-05-07:
@tumbili have you inspected the rate limiting? Why are we seeing the spikes in @Zefz log?

tumbili, 2015-05-07:
There are spikes but what do they have to do with rate limiting?

tumbili, 2015-05-07:
Maybe we are talking of different things

LorenzMeier, 2015-05-07:
actually at 360 dps -4 rad/s is possible. The question is if 360 dps is the right rate limit as default. Maybe we should apply a tighter limit?

tumbili, 2015-05-07:
No, the problem is something else. Why is there a spike in the roll_rate_sp? There is not huge attitude error.

tumbili, 2015-05-07:
There is also a huge spike in the pitch setpoint

LorenzMeier, 2015-05-07:
Yes, those spikes are what I'm trying to hunt down. Do they happen in your flight log as well?

tumbili, 2015-05-07:
No

tumbili, 2015-05-07:
And I see that attitude setpoint quaternions are not logged, probably the new estimator does not log yet.

tumbili, 2015-05-07:
Because that is the relevant thing to look at.

tumbili, 2015-05-07:
Just fixing that

tumbili, 2015-05-07:
Ah, I had that mistaken. Ok, I see that the quaternion setpoint is generally not logged.

Zefz, 2015-05-07:
Here is another log showing the same problem. It enters PosHold first and AltHold the second time.

It seems the attitude spike setpoints only happen on the first initial mode change into a flight assisted mode. Curiously this did not seem to happen without GPS (AltHold only). Maybe this is a problem in the mc position controller?

tumbili, 2015-05-07:
@Zefz @LorenzMeier A possible cause for the spike seem to be the desired acceleration in y direction of the position controller. If you plot it in flight plot you can see it very nicely. I just haven't understood yet where this spike comes from.

tumbili, 2015-05-07:
@Zefz @LorenzMeier One cause I see is that the mc_pos_controller does not reset it's D part.

kd0aij, 2015-05-07:
It looks to me as though RC inputs are commanding the spikes at around 43.5 seconds. Am I looking at the wrong event?
[image]

tumbili, 2015-05-07:
@kd0aij I'm looking at t == 16 seconds

kd0aij, 2015-05-07:
will do a quick POSCTL check with Q estimator

tumbili, 2015-05-07:
@kd0aij Just a second, can you use the PR I will do in 10 seconds?

kd0aij, 2015-05-07:
OK

kd0aij, 2015-05-07:
checked out branch mc_pos_D_reset. flashing now

kd0aij, 2015-05-07:
very short flight; battery low. But it appeared to start toilet-bowling immediately on entering POSCTL.
Looks like mag calibration problems again.

tumbili, 2015-05-07:
@kd0aij Could you show the log please?

tumbili, 2015-05-07:
@kd0aij I did a param reset today with my IRIS and flew position control mode. Everything worked fine.
That's why I'm really surprised.

kd0aij, 2015-05-07:
http://dash.oznet.ch/view/9vTwVDUV88TTTkQE8GGBJS
Toilet bowling strong about 8 seconds after entering posctl

kd0aij, 2015-05-07:
Interestingly, no problem with Flow: http://dash.oznet.ch/view/8bYbgkFh8DzR3stnTEKgNU
Back to GPS, and posctl is still no good: http://dash.oznet.ch/view/6hPRkZutbLuWfUAMTQZreE

LorenzMeier, 2015-05-08:
@kd0aij Thrust and magnetic field are correlated in your case:
http://dash.oznet.ch/view/6hPRkZutbLuWfUAMTQZreE#Thrust_PLOT

The overall vector orientation is different between power-off and power on (look at the takeoff sequence):
http://dash.oznet.ch/view/6hPRkZutbLuWfUAMTQZreE#IMU_Mag_PLOT

To me it looks marginal - it might just work or just not work, very hard to tell. It would certainly be enough to explain toilet-bowling, but it its not as strong as I could definitely pinpoint it as the source.

LorenzMeier, 2015-05-08:
And in your last log with GPS, your roll / pitch tracking is horrible:
http://dash.oznet.ch/view/6hPRkZutbLuWfUAMTQZreE#A_PitchRate_PLOT

This is your primary issue. As long as that is not resolved this quad will continue to raise false positives in terms of performance. You really, really need to re-tune it 8).

LorenzMeier, 2015-05-08:
I'm closing this in favour of #2143, which likely will resolve this.

closed 2015-05-08 by LorenzMeier

kd0aij, 2015-05-08:
@LorenzMeier This is very interesting... How could the pitchrate tracking be so good in the flow run, and so bad on the next flight with GPS a minute later? I certainly didn't change any tuning parameters, just put the lens cap on the flow camera.

LorenzMeier, 2015-05-08:
There are also suboptimal areas in the manual flight section of the flow log. Your system seems to be right at the edge of ok, but not quite there.

kd0aij, 2015-05-08:
Any suggestions on what might be wrong with the tuning? In my opinion there
are no problems with manual flight so the required procedure is completely
unknown to me.

kd0aij, 2015-05-08:
I guess this points out how valuable an auto-tuning procedure would be. It would encapsulate the knowledge which I (the user) lack and improve user experience immeasurably by eliminating crashes which currently result simply from switching from manual to posctl mode. A step in the right direction would be an auto-analysis procedure which examines a log and suggests changes to specific tuning parameters to correct problems.