PX4/PX4-Autopilot #2144: EKF altitude estimation - measurement noise of baro

surberj, 2015-05-07:
Do we have a parameter capturing the measurement noise variance of the barometer? 
As far as I understand:
Regarding estimation of the altitude, the IMU is used for propagation and GPS and Baro measurements are used for the update. For the GPS-Down measurement the parameters PE_POSD_NOISE and PE_VELD_NOISE are used. There should be as well a parameter capturing the variance of the baro measurement noise. Can someone help me with that?

Regarding this issue there exists also a parameter PE_GPS_ALT_WGT, which is currently nowhere used in the code.

Zefz, 2015-05-08:
PE_POSD_NOISE is for barometer. Only GPS-Down-Velocity is used, GPS-Down-Position is not used for altitude estimations. (only for long-term barometer drift, but that is not a parameter that you can tweak)

LorenzMeier, 2015-05-08:
Please ask questions on the mailing list: http://groups.google.com/group/px4users

Thanks!

closed 2015-05-08 by LorenzMeier