PX4/PX4-Autopilot #2145: PX4Flow not detected on boot

kd0aij, 2015-05-07:
I have PX4FLow connected to Pixhawk via I2C.
Everything is fine on a cold boot initiated by plugging in USB.
But on a cold boot from battery/power module, the flow module is not detected.
Inserting a 4 second delay into rc.sensors prior to "px4flow start" fixed the problem (3 seconds is not enough).

davids5, 2015-05-07:
Hi @kd0aij, 

This may be related to https://github.com/PX4/Firmware/issues/2001#issuecomment-93524623

The HW should be here next week for me to test.

kd0aij, 2015-05-07:
Thanks David. Wish I had seen that a few days ago!

pkocmoud, 2015-05-07:
David, I tried to order the FLOW thru 3DR and it was back-ordered for who knows how long. I bought 2 from the UK distributor. Let me know if you need to borrow one. I can ship one out to you today.

davids5, 2015-05-07:
@pkocmoud Thank you I may need to take you up on the offer. I am checking on an ETA and will let you know.

davids5, 2015-05-08:
@pkocmoud Thank you again, but the hw is on it's way here.

LorenzMeier, 2015-05-08:
I suspect the quick fix would be to have the px4flow driver try for a few seconds before it gives up. However, understanding the power sequencing would be great.

davids5, 2015-05-11:
The board arrived - Duh- I forgot to ask for the j-tag connector to be added. I have some Samtec .05 ordered and will look at this as soon as I can debug it.

Would somebody please describe the cabling setup. A photo would work.

David

kd0aij, 2015-05-11:
I just followed these setup instructions:
https://pixhawk.org/modules/px4flow
and connected px4flow I2C to pixhawk I2C.

Just be sure you orient px4flow Y axis toward front of vehicle, or change
*SENS_FLOW_ROT* to reflect the actual orientation.

davids5, 2015-05-19:
@kd0aij @pkocmoud @LorenzMeier 

I have tested with PX4FMUV2 with ONLY the power module connected.  Test was conducted by disconnecting my bench power supply.

PX4 Firmware c06ba047e2ded236b8ce5e17694dfca8b8d6f067
With 
1) Old PX4 flow bootloader - https://github.com/PX4/Flow/blob/master/px4flow_bl.elf
2) Current  Bootloder https://github.com/PX4/Bootloader/commit/a8c0160397062d1760f35ef8eda84ef0db709e63
3) New Bootloader https://github.com/PX4/Bootloader/tree/vetting

I can not reproduce a failure. By Failure I an referring to not recieiving  [PX4FLOW] on I2C bus 1 at 0x42 (bus: 100 KHz, max: 400 KHz) from this [issue]  (https://github.com/PX4/Firmware/issues/2001#issuecomment-91478230) 

There was a bug in the booloader (built after Feb 2014 and before  Mar 27 2015 - on the current master) that could have caused AN issue - a boot without USB connected could hang in the uab library code.  But since a delay fixed the issue as reported by   @h46incon I do not think that was the root cause.

Was there a change in any startup script timing?

davids5, 2015-05-19:
@kd0aij - Would you be able to re-test with current PX4 Firmware at c06ba047e2ded236b8ce5e17694dfca8b8d6f067 and see if the problem still exists. If So the I would like you to try the Boorloader from  at https://github.com/PX4/Bootloader/tree/vetting fdc247e and let me know if it fixes the issue.

kd0aij, 2015-05-19:
**problem still present with latest master**
nsh> ver all                                                                    
FW git-hash: c06ba047e2ded236b8ce5e17694dfca8b8d6f067                           
Build datetime: May 18 2015 19:58:49                                            

**cold boot with USB power:**
```
sercon: Registering CDC/ACM serial driver                                  
sercon: Successfully registered the CDC/ACM serial driver                       
[i] microSD mounted: /fs/microsd                                                
nshterm [6:70]                                                                  
param: selected parameter default file /fs/mtd_params                           
[param] Loaded: /fs/mtd_params                                                  
[rgbled] on I2C bus 2 at 0x55 (bus: 100 KHz, max: 100 KHz)                      
nsh: pca8574: command not found                                                 
 1:   SYS_USE_IO: match                                                         
 4010: + SYS_AUTOSTART: match                                                   
px4io: CRCs match                                                               
dataman: Power on restart, data manager file '/fs/microsd/dataman' size is 1015s
[MS5611_SPI] on SPI bus 4 at 2 (11000 KHz)                                      
ms5611: no device on bus 4                                                      
[MS5611_SPI] on SPI bus 1 at 3 (11000 KHz)                                      
ms5611: no device on bus 1                                                      
ms5611: no device on bus 2                                                      
<adc> init done                                                                 
ver hwcmp match: PX4FMU_V2                                                      
hmc5883: no device on bus 2                                                     
hmc5883: no device on bus 1                                                     
mpu6000: driver start failed                                                    
[MPU6000] on SPI bus 1 at 4 (1000 KHz)                                          
[L3GD20] on SPI bus 1 at 1 (11000 KHz)                                          
[LSM303D] on SPI bus 1 at 2 (11000 KHz)                                         
meas_airspeed: no MS4525 airspeed sensor connected                              
ets_airspeed: no ETS airspeed sensor connected                                  
ets_airspeed: no ETS airspeed sensor connected      
**** [PX4FLOW] on I2C bus 1 at 0x42 (bus: 100 KHz, max: 400 KHz)*******************
telem> PREFLIGHT FAIL: MAG #0 UNCALIBRATED                                      
telem> Not ready to fly: Sensors need inspection                                
commander: INVAL: ARMING_STATE_STANDBY_ERROR - ARMING_STATE_STANDBY             
[px4io] default PWM output device                                               
mavlink_if0: mode: 0, data rate: 1000 B/s on /dev/ttyS1 @ 57600B                
ver hwcmp match: PX4FMU_V2                                                      
[i] MULTICOPTER                                                                 
+ MAV_TYPE: unchanged                                                           
[px4io] mixer sent                                                              
[i] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0                      
 1: + INAV_ENABLED: match                                                       
attitude_estimator_q: started                                                   
.                                                                               
[i] Addons script: /fs/microsd/etc/extras.txt                                   
sdlog2: log buffer size: 22528 bytes                                            
waiting 4 seconds for PX4Flow to boot                                           
navigator: Could not clear geofence                                             
navigator: offboard mission updated: dataman_id=0, count=0, current_seq=0       
mavlink_if0: DISABLING HARDWARE FLOW CONTROL                                    
telem> settings autosaved                                                       
position_estimator_inav: baro offs: 1582                                     
****px4flow: already started ****************************************************************
                                                                               
NuttShell (NSH)                                                                 
nsh> 
```

**cold boot from battery:**
```
sercon: Registering CDC/ACM serial driver                                  
sercon: Successfully registered the CDC/ACM serial driver                       
[i] microSD mounted: /fs/microsd                                                
nshterm [6:70]                                                                  
param: selected parameter default file /fs/mtd_params                           
[... 26 lines cut ...]
commander: INVAL: ARMING_STATE_STANDBY_ERROR - ARMING_STATE_STANDBY             
[px4io] default PWM output device                                               
mavlink_if0: mode: 0, data rate: 1000 B/s on /dev/ttyS1 @ 57600B                
ver hwcmp match: PX4FMU_V2                                                      
[i] MULTICOPTER                                                                 
[... cut ...]
mavlink_if0: DISABLING HARDWARE FLOW CONTROL                                    
telem> settings autosaved                                                       
position_estimator_inav: baro offs: 1583                                        
[PX4FLOW] on I2C bus 1 at 0x42 (bus: 100 KHz, max: 400 KHz)      
****PX4Flow start succeeded*********************************************************************
                                                                                
NuttShell (NSH)                                                                 
nsh> 
```

kd0aij, 2015-05-19:
@davids5 
The startup messges above show that the Flow module is not detected without a 4 second pause when booting from battery.
Do you mean to replace the bootloader in the FMU or the PX4Flow?
There are no instructions for the Flow module... will it require installing a JTAG connector?

davids5, 2015-05-19:
@kd0aij Thank you. I guess we need to understand the difference between our two systems. Can you sent me your parameters, any diffs to scripts  and a list of what is connected to what io.

I only have debug port, CAN 1 and the px4 flow connected.

I was speaking of the px4flow's bootloader - yes you have to add the connector to use the jtag (I can Send you one if need be) 

I'm on a mobile device, I will check the schematics tomorrow and see if there's a way to do it over the serial port using BOOT0/1

kd0aij, 2015-05-19:
The only change I've made to startup is to add extras.txt:
```
nsh> cat /fs/microsd/etc/extras.txt                                             
# Check for flow sensor                                                         
echo "waiting 4 seconds for PX4Flow to boot"                                    
sleep 4                                                                         
if px4flow start                                                                
then                                                                            
        echo "PX4Flow start succeeded"                                          
fi                                                                              
```
The PX4Flow and external mag are connected via I2C splitter, but behavior is the same with external mag disconnected.
Here is the parameter file: https://copy.com/ZaRALrEZUJqGX6Ow
I hope the bootloader doesn't need to change (or that it can by upgraded over USB), so existing users won't have to buy a JTAG probe and solder a connector.

pkocmoud, 2015-05-19:
Mark, can you query the board for the boot loader version you have?

kd0aij, 2015-05-19:
Don't know how to do that...

kd0aij, 2015-05-19:
@davids5 BTW, USB is not connected to my PX4Flow in normal operation, just I2C

davids5, 2015-05-19:
@kd0aij 
[POWERMOD]
     |
[FMUV2]----i2c---[PX4FLOWV1]
     |_> > - FTDI debug cable (1)
     |_>> - FTDI debug cable (2)

So the only connections at power application are Power Module. PX4FLOW with 12" I2C cable and Arming Switch. Then I add the FTTDI debug cable.

The trace below was with your parameters downloaded from GCS then all USB disconnected
and the FTDI cable unplugged - connect power 6.0 V lab supply to the POWER module. Then plug in the FTDI cable. (Just to prevent back feed powering) 
```
þ=f¸sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
[i] microSD mounted: /fs/microsd
nshterm [6:70]
param: selected parameter default file /fs/mtd_params
[param] Loaded: /fs/mtd_params
[rgbled] on I2C bus 2 at 0x55 (bus: 100 KHz, max: 100 KHz)
nsh: pca8574: command not found
 1:   SYS_USE_IO: match
 4010: + SYS_AUTOSTART: match
px4io: CRCs match
dataman: Power on restart, data manager file '/fs/microsd/dataman' size is 101504 bytes
[MS5611_SPI] on SPI bus 4 at 2 (11000 KHz)
ms5611: no device on bus 4
[MS5611_SPI] on SPI bus 1 at 3 (11000 KHz)
ms5611: no device on bus 1
ms5611: no device on bus 2
<adc> init done
ver hwcmp match: PX4FMU_V2
hmc5883: no device on bus 2
hmc5883: no device on bus 1
mpu6000: driver start failed
[MPU6000] on SPI bus 1 at 4 (1000 KHz)
[L3GD20] on SPI bus 1 at 1 (11000 KHz)
[LSM303D] on SPI bus 1 at 2 (11000 KHz)
meas_airspeed: no MS4525 airspeed sensor connected
ets_airspeed: no ETS airspeed sensor connected
ets_airspeed: no ETS airspeed sensor connected
[PX4FLOW] on I2C bus 1 at 0x42 (bus: 100 KHz, max: 400 KHz)
telem> PREFLIGHT FAIL: MAG #0 UNCALIBRATED
telem> Not ready to fly: Sensors need inspection
commander: INVAL: ARMING_STATE_STANDBY_ERROR - ARMING_STATE_STANDBY
[px4io] default PWM output device
mavlink_if0: mode: 0, data rate: 1000 B/s on /dev/ttyS1 @ 57600B
ver hwcmp match: PX4FMU_V2
[i] MULTICOPTER
+ MAV_TYPE: unchanged
[px4io] mixer sent
[i] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0
 1: + INAV_ENABLED: match
attitude_estimator_q: started
.

NuttShell (NSH)
nsh> navigator: Could not clear geofence
navigator: offboard mission updated: dataman_id=0, count=0, current_seq=0
mavlink_if0: DISABLING HARDWARE FLOW CONTROL
sdlog2: log buffer size: 22528 bytes
telem> settings autosaved
position_estimator_inav: baro offs: 117
ver all
HW arch: PX4FMU_V2
FW git-hash: c06ba047e2ded236b8ce5e17694dfca8b8d6f067
Build datetime: May 18 2015 13:26:56
Toolchain: 4.8.4 20140725 (release) [ARM/embedded-4_8-branch revision 213147]
MCU: STM32F42x, rev. A

WARNING   WARNING   WARNING!
Revision A has a silicon errata
This device can only utilize a maximum of 1MB flash safely!
http://px4.io/help/errata

UID: 26001A:34324708:31323533
```

So what is different?

What model number is your Power Module?
What is the Voltage of your Batteries into the PM?

kd0aij, 2015-05-19:
the power module is the one that comes with the Pixhawk; I can't see a model number on it.
I'm using both 3S and 4S LiPo packs on this quad, 3S for these tests.
Are you using the latest master PX4Flow firmware aad67ed ?

davids5, 2015-05-19:
@kd0aij Can you measure the voltage into the PM?

I followed you original setup using GCS and it flashed the board. Not Sure of the version .

But I just built Flow master locally and and downloaded aad67ed, Same behaviour as before.

I am not convinced it is that bootloader. But lets rule it out - Try this: 

Option 1) Can you solder? - If So supply the USB V5 (pin1) from the i2c (pin1) with a jumper wire.
Option 2) Plug the flow into a Phone Charger at the same time as the PM is connected. 

I can not see a definitive way to determine the version of the bootloader. On Windows you can run usbview and see if the descriptors are messed up. That would represent the bootloader that is in master as a bin file (I was calling old)

kd0aij, 2015-05-19:
OK, I've also demonstrated the problem flashing via QGC and make upload.

I'm not sure why connecting USB pin1 to I2C pin1 is of interest, since
there's nothing connected to the Flow's USB port in either test scenario.
In both cases, the Flow is powered via I2C;
But maybe you mean connecting the FMU's USB pin1 to the Flow's I2C pin1...
I'll just check whether I2C power is delayed when booting from battery.

davids5, 2015-05-19:
@kd0aij are you on Skype? Please call me david_s5

davids5, 2015-05-19:
[SOLVED] Sort - of

With @kd0aij we have determined that his flow does not skip the 5 seconds of boot delay on his hardware **even when powered off of ONLY i2c**. 

We do not know the cause as of yet. It might be an older bootloader.  You can simply determine if you have the issues by powering the Flow only via I2C and observer the orange LED. If it blinks rapidly for ~ 5 Seconds -  the PX4Flow is entering Bootloader mode, at power on, and you you will need to use  Mark's fix
```
cat /fs/microsd/etc/extras.txt
# Check for flow sensor
echo "waiting 4 seconds for PX4Flow to boot"
sleep 4
if px4flow start
then
        echo "PX4Flow start succeeded"
fi
```

We will determine if a bootloader upgrade will solve the issues and determine the best way to deploy the new Bootloader.

David

LorenzMeier, 2015-05-20:
@kd0aij A change to the driver would be welcome which exits the start handler with return code 0, tries to connect for 5 seconds and eventually gives up. I don't want to put the sleep solution which holds up the overall boot into master though.

kd0aij, 2015-05-20:
I'm unable to see the PX4Flow with the BlackMagic probe and gdb "mon swdp_scan" command.
Is the procedure the same as for PX4FMU and IO?

davids5, 2015-05-20:
Hi @kd0aij 

Yes

Here is what I get.
top                                           BOTTOM
[BM]-------CABLE-----------------[PX4FLOW]
 ^Pin 1 (red)------------------------^Pin 1 (red) 
```
01:35:39 {vetting} ~/mainline/Bootloader$ arm-none-eabi-gdb px4flow_bl.elf
GNU gdb (GNU Tools for ARM Embedded Processors) 7.6.0.20140731-cvs
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=i686-linux-gnu --target=arm-none-eabi".
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>...
Reading symbols from 
mainline/Bootloader/px4flow_bl.elf...done.
(gdb) target extended-remote /dev/ttyACM0
Remote debugging using /dev/ttyACM0                                                                                                                                                                                                                                     
(gdb)  mon swdp_scan                                                                                                                                                                                                                                                    
Target voltage: 3.3V                                                                                                                                                                                                                                                    
Available Targets:                                                                                                                                                                                                                                                      
No. Att Driver                                                                                                                                                                                                                                                          
 1      STM32F4xx                                                                                                                                                                                                                                                       
(gdb) mon                                                                                                                                                                                                                                                               
Black Magic Probe (Firmware 1.5-00279-gfd9eef8, build 20141028)                                                                                                                                                                                                         
Copyright (C) 2011  Black Sphere Technologies Ltd.                                                                                                                                                                                                                      
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>    
```

kd0aij, 2015-05-21:
Then I have to conclude there's something wrong with my PX4Flow module, or give up trying to solder JTAG connectors :)  But they work on my FMUv1 and PX4IO boards... 2 out of 3 ain't bad, I guess.

LorenzMeier, 2015-06-25:
Update: This is a problem in some flow boards and their boot loader. We need a workaround for the firmware driver.

LorenzMeier, 2015-07-08:
This is fixed on master and beta, closing.

closed 2015-07-08 by LorenzMeier

tangentaudio, 2015-08-07:
I'm experiencing a possibly related issue.  I just got my PX4FLOW the other day and I updated it to the latest stable release via QGC.  My Pixhawk is also updated to latest stable also via QGC.  I can talk to the PX4FLOW direct via USB and get an image in QGC, and see the variables change as expected in the analyze tab.

If I plug it into the Pixhawk via an I2C hub, along with the GPS magnetometer, and power the system up (either via USB or via battery) - neither the PX4FLOW or the mag work.  I see the LEDs on the PX4FLOW board blink quite dimly a few times, then they stop.  I haven't had a chance yet to put a scope on SDA/SCL to see the state of the I2C bus.

However, if I boot it up with just the GPS mag connected to the I2C bus and boot it up, it works as expected.  Once booted, if I then hot-plug the PX4FLOW onto the I2C bus and then do 'px4flow start' from nsh, it detects it and seems to work just fine.

Any thoughts?  It's a brand new PX4FLOW direct from 3DR, so I'd hope it would be up to date in terms of bootloader, etc.

NCharabaruk, 2016-09-29:
I'm having the same issue as tangentaudio. Has this ever been fixed?

davids5, 2018-10-10:
@rhamrah the console. See https://dev.px4.io/en/debug/system_console.html

rhamrah, 2018-10-17:
> @rhamrah the console. See https://dev.px4.io/en/debug/system_console.html

Thanks David, it really helped. Could you please let me know of your ideas/comments on the question I asked in #10671 ?