PX4/PX4-Autopilot #2148: Stale git version on incremental builds

AndreasAntener, 2015-05-08:
If you build the firmware and only do incremental builds, the commit SHA that's later used in sdlog2 won't change.

dagar, 2015-05-08:
It's passed as a define. I'm not sure how to make the current build system pick up on that change. Maybe we could write it to a header (and setup header deps) or another c file?

AndreasAntener, 2015-05-08:
Just quoting @kd0aij , might be an option
> we should add a "touch" to the makefile to force recompiling sdlog2 and ver

kd0aij, 2015-05-08:
I'm pretty inept with "make", but this caused sdlog2.c to be recompiled:
$(shell touch src/modules/sdlog2/sdlog2.c)
I inserted it in the makefile immediately following "export GIT_DESC"

kd0aij, 2015-05-08:
created PR #2149 with makefile change

LorenzMeier, 2015-05-24:
This is fixed on master. Thanks @kd0aij for your PR, fixed it in a similar fashion.

closed 2015-05-24 by LorenzMeier