PX4/PX4-Autopilot #2166: corrupted parameter values in ROM

kd0aij, 2015-05-13:
I recently had INAV_ENABLED change (unexpectedly) from 1 to 0 in a Pixhawk which I had set up to run the Q estimator and INAV. This correlated with flow/POSCTL becoming unstable. Changing INAV_ENABLED back to 1 using Tower made flow based POSCTL work again, but the logfile from that flight shows "INAV_ENABLED 1.06535322E9". Rebooting the Pixhawk showed the ekf_att_pos_estimator running... no att_estimator_q or INAV.

LorenzMeier, 2015-05-13:
Tower is corrupting the values. Please file an issue against it. It's a known issue as its not adhering to MAVLink specs.

closed 2015-05-13 by LorenzMeier

kd0aij, 2015-05-13:
@LorenzMeier Could you point me to the known issue with Tower violating the mavlink spec?

LorenzMeier, 2015-05-13:
APM is not interpreting params as per the standard for non-float values. They would need to adjust their param conversion routines depending on which autopilot they are connected to. If you mention me in the issue I can provide the link.

kd0aij, 2015-05-13:
@LorenzMeier BTW, this seems to indicate a lack of input validation on the PX4 side, since 1E9 is somewhat larger than the spec'd max value for the parameter.

kd0aij, 2015-05-13:
@LorenzMeier https://github.com/DroidPlanner/Tower/issues/1483

LorenzMeier, 2015-05-13:
There is and will be no input validation. It's pointless as its the inner working of the system and there is a million ways to crash a drone using individually correct, but collectively wrong parameters. At the point where we can validate them collectively we have built a system that doesn't need parameters anymore - that would be a great end goal though.

LorenzMeier, 2015-05-13:
We might be able to add range limits cheaply which would have prevented this, to me that's not real input validation though. Is that what you meant?

kd0aij, 2015-05-13:
yes, the valid range of INAV_ENABLE is [0,1]
I presume that PX4 behavior is "undefined" if it's not in that range

kd0aij, 2015-05-13:
by "valid range" I mean what I read in the parameter description