PX4/PX4-Autopilot #2172: Add File Request from GCS

davids5, 2015-05-13:
To support fetching firmware for devices.

dagar, 2017-04-11:
Still interested in this?

pavel-kirienko, 2017-04-11:
Yes.

dagar, 2017-04-11:
Can you expand on the idea a bit? The firmware sends a request to the GCS to download new firmware which is then sent to the autopilot?

davids5, 2017-04-11:
The UAVCAN fw can be fetched from a well known URL. It then needs to be uploaded to the SD card as defined [here](https://github.com/mavlink/mavlink/pull/414).

davids5, 2017-04-11:
The above reference may be a tad stale. 
The [uavcan_servers](https://github.com/PX4/Firmware/blob/master/src/modules/uavcan/uavcan_servers.cpp#L870) is a better reference - It supports a copy from ROMFS to the sd card. There was an addition of a leading underscore to ID the ROM FS version.

closed 2018-02-22 by PX4BuildBot