PX4/PX4-Autopilot #2179: get usart6 on stm32f4discovery working

nopeppermint, 2015-05-15:
@LorenzMeier @davids5 
I got usart6 on stm32f4discovery with current master working (which contains already #2174), but there are changes in the Nuttx submodule necessaries in the file Nuttx/nuttx/configs/stm32f4discovery/include/board.h

So I have to provide a PR to the Nuttx submodule right? 

```

Parent: 678eea3d2c157c686439597abb0367b0f1e643f4 (Merge pull request #57 from PX4/master_usb_fix)

---------------- nuttx/configs/stm32f4discovery/include/board.h ----------------
index de829f1..fb8ed94 100644
@@ -211,16 +211,27 @@
  * The STM32F4 Discovery has no on-board serial devices, but the console is
  * brought out to PA2 (TX) and PA3 (RX) for connection to an external serial device.
  * (See the README.txt file for other options)
  */
 
 #define GPIO_USART2_RX GPIO_USART2_RX_1
 #define GPIO_USART2_TX GPIO_USART2_TX_1
 
+/* UART6:
+ *
+ * The STM32F4 Discovery has no on-board serial devices, PC6 (TX) and PC7 (RX) 
+ * for connection to an external serial device.
+ */
+
+#define GPIO_USART6_RX GPIO_USART6_RX_1
+#define GPIO_USART6_TX GPIO_USART6_TX_1
+
+/* USART6 require a RX DMA configuration */
+#define DMAMAP_USART6_RX DMAMAP_USART6_RX_1
 /* PWM
  *
  * The STM32F4 Discovery has no real on-board PWM devices, but the board can be
  * configured to output a pulse train using TIM4 CH2 on PD13.
  */
 
 #define GPIO_TIM4_CH2OUT GPIO_TIM4_CH2OUT_2

```

LorenzMeier, 2015-05-16:
You have to send this as PR to PX4/NuttX.

nopeppermint, 2015-05-17:
I just updated my local copy of the repo, recompiled everything (including nuttx) and retested the px4-stm32f4discovery usart6 capability and can now confirm that px4/master @https://github.com/PX4/Firmware/commit/3032afb80399a78fe56417858d7b026db8206634 has a fully working usart6 / ttyS1 on stm32f4discovery as you can see below. 

So there won't be a PR against px4/nuttx.

the Wiki is updated https://pixhawk.org/modules/stm32f4discovery

@LorenzMeier Sorry for confusing https://gitter.im/PX4/Firmware?at=5554dca0b2001f92677cb408
I didn't recompiled everything, so I wasn't able to see the new usart6 :-|

[image]

closed 2015-05-17 by nopeppermint