PX4/PX4-Autopilot #2184: MAVLink FTP file listing fails on some data inputs

LorenzMeier, 2015-05-15:
@DonLakeFlyer Logging here my WIP findings. I used the image of Roman's card to recreate his setup. It reproduces 100% for me.

If I start MAVLink via QGC (via rc.usb), I get a NACK on the file listing after some files. If I start MAVLink on the shell like this:
```
mavlink start -r 80000 -d /dev/ttyACM0 -x
```
I get a timeout. This is interesting because it shouldn't matter on which shell I start the app. Looks like a first hint or an orthogonal bug.

LorenzMeier, 2015-05-16:
Ok, mystery solved. The shell was still sitting on the port, which is why RX failed. Now I get this:

```
mavlink_rcv_if1: FTP: list /fs/microsd/.Trash-1000/files/2011-02-20 offset 0
FTP: list /fs/microsd/.Trash-1000/files/2011-02-20 D2011-02-20
FTP: list /fs/microsd/.Trash-1000/files/2011-02-20 Dsess001
FTP: list /fs/microsd/.Trash-1000/files/2011-02-20 Dsess002
FTP: list /fs/microsd/.Trash-1000/files/2011-02-20 Dsess003
FTP: list /fs/microsd/.Trash-1000/files/2011-02-20 Dsess004
FTP: list /fs/microsd/.Trash-1000/files/2011-02-20 Dsess005
mavlink_rcv_if1: FTP: Ack seq_number: 194
mavlink_rcv_if1: FTP: received ftp protocol message target_system: 1
ftp: channel 1 opc 3 size 40 offset 6
mavlink_rcv_if1: FTP: can't open path '/fs/microsd/.Trash-1000/files/2011-02-20'
mavlink_rcv_if1: FTP: Nak seq_number: 196
```
And manually listing the directory yields:
```
ls: no such directory: /fs/microsd/.Trash-1000/files/2011-02-20/
```
Or in other words: This is a filesystem issue.

LorenzMeier, 2015-05-16:
@DonLakeFlyer I pushed an attempt to fix and a workaround to master. I don't understand the state machine fully yet, but what I'm trying to return is an empty directory. QGC is keeping requesting the directory. I would appreciate if  you could have a look. If you remove the first condition and flash Roman's microSD image you should get an endless loop of directory requests.
https://github.com/PX4/Firmware/blob/2432418751b341aae150bccb275a7908ae107ee7/src/modules/mavlink/mavlink_ftp.cpp#L376

DonLakeFlyer, 2015-05-18:
Ah, makes sense. QGC is confused by the EOF logic and an empty directory. Easy fix on QGC side.

LorenzMeier, 2015-05-19:
This is fixed, thanks!

closed 2015-05-19 by LorenzMeier