PX4/PX4-Autopilot #2188: genromfs no longer available?

asakusuma, 2015-05-17:
Following the [mac installation instructions](https://pixhawk.org/dev/toolchain_installation_mac), a couple of the brew formulas don't appear to exist. I'm running Mac OS 10.10 and Homebrew 0.9.5.
```
~/workspace/Firmware on master*
$ brew install genromfs
Error: No available formula for genromfs 
Searching formulae...
Searching taps...
~/workspace/Firmware on master*
$ brew install kconfig-frontends
Error: No available formula for kconfig-frontends 
Searching formulae...
Searching taps...
```

Is there somewhere else I can find genromfs and kconfig-frontends for mac? Some google searching seems to indicate that both are primarily linux tools. The lack of genromfs is, I assume, causing the following error when I they try and make:

```
$ make -j8 px4fmu-v2_default
...
Pruning ROMFS files.
ROMFS:   romfs.img
/bin/sh: genromfs: command not found
make[1]: *** [romfs.img] Error 127
make[1]: *** Waiting for unfinished jobs....
Running from the source directory
%  MODULE_NAME         = uavcan
%  MODULE_SRC          = /Users/akusuma/workspace/Firmware/src/modules/uavcan/
%  MODULE_OBJ          = /Users/akusuma/workspace/Firmware/Build/px4fmu-v2_default.build//Users/akusuma/workspace/Firmware/src/modules/uavcan/module.pre.o
%  MODULE_WORK_DIR     = /Users/akusuma/workspace/Firmware/Build/px4fmu-v2_default.build//Users/akusuma/workspace/Firmware/src/modules/uavcan
CMD:     uavcan
make: *** [/Users/akusuma/workspace/Firmware/Build/px4fmu-v2_default.build/firmware.px4] Error 2
```

LorenzMeier, 2015-05-17:
The formulas come from here: https://github.com/PX4/homebrew-px4. You forgot to tap the PX4 repo, please re-read the installation instructions to resolve your issue.

closed 2015-05-17 by LorenzMeier

asakusuma, 2015-05-17:
@LorenzMeier I did not forget to tap the repo. But just to be sure, I tried running through the instructions again and, sure enough:
```
~/workspace/Firmware on master*
$ brew tap PX4/homebrew-px4
Warning: Already tapped!
~/workspace/Firmware on master*
$ sudo brew update
Password:
Updated Homebrew from e4e6da9e to b253cd19.
==> New Formulae
pkgdiff
==> Updated Formulae
abcm2ps		  cpputest	    iojs	      nginx		radare2		  speedtest_cli	    sqsh	      ssdeep
abcmidi		  heroku-toolbelt   libghthash	      paperkey		sonar-runner	  sphinx	    ssdb
~/workspace/Firmware on master*
$ brew install genromfs
Error: No available formula for genromfs 
Searching formulae...
Searching taps...
```

asakusuma, 2015-05-17:
@LorenzMeier figured out the issue, had to untap and then re-tap with sudo

```
Warning: Could not create link for px4/px4/genromfs, as it
conflicts with Homebrew/homebrew/genromfs. You will need to use the
fully-qualified name when referring this formula, e.g.
  brew install px4/px4/genromfs
```

LorenzMeier, 2015-05-17:
You should not be needing sudo for homebrew and its actively discouraged. Your homebrew installation is most likely in a bad shape and you might have further issues down the road (try brew doctor).