PX4/PX4-Autopilot #2192: Board rotation offsets - Bug in matrix multiplication?

yyassa, 2015-05-18:
The SENS_BOARD_*_OFF parameters for fine-tuning the board rotation are multiplied by the rough board rotation in two in different ways (in board frame vs aircraft body frame) in different parts of the code:

accelerometer_calibration.cpp (line 433):
```
	/* combine board rotation with offset rotation */
	board_rotation = board_rotation_offset * board_rotation;
```

sensors.cpp (line 859):
```
_board_rotation = _board_rotation * board_rotation_offset;
```

Which one should it be?

tumbili, 2015-05-18:
The first one looks better in my opinion. It's true that rotation matrix multiplication does not commute.
I guess the large board rotation should be before the fine tuning.

kd0aij, 2015-05-18:
I concur:
If board_rotation transforms from board_frame to body_frame1 and
board_rotation_offset transforms from body_frame1 to body_frame2
then (board_rotation_offset * board_rotation) transforms from board_frame to body_frame2.
In that case, the rotation offsets are measured from body_frame1 to body_frame2, which I think is the way they are intended.

LorenzMeier, 2015-05-19:
@tumbili Could you please resolve this? Thanks!

tumbili, 2015-05-19:
Fixed via #2197

closed 2015-05-19 by LorenzMeier