PX4/PX4-Autopilot #2193: Level / Board rotation method

LorenzMeier, 2015-05-18:
@DonLakeFlyer @tumbili As discussed on the dev call it seems we need an option for users to "level" a system during calibration. That would be a fourth button in the sensor config view.

The instructions should be something along the lines of: Put the system on a perfectly level surface, then click ok. The system will sample the current attitude for a very short amount of time and set a number of params. Levelling is not strictly required, so not sure about the indicator color.

kd0aij, 2015-05-18:
It would also help to explain that the current accel calibration does not rely on the upright orientation being precisely level, and that the other 5 orientations also need not be precise. 
The new "level" function would refine the autopilot rotation matrix (this is not currently being performed).

DonLakeFlyer, 2015-05-19:
If you don't do the extra leveling step, then you need to make sure you are level when you do upright regular accel cal don't you? And that your board rotation exactly matches one of the selections? You only need to perform this extra step if your selected board orientation wouldn't bring things to exact level?

Or in other words how do you explain to the user when they would NOT need to perform the level step?

For a fixed wing tail dragger, what is level? I would assume level flight, not level with the tail wheel down.

kd0aij, 2015-05-19:
What I tried to say above is that you currently don't need to be precisely level when you do "upright regular accel cal" (try it, it's pretty cool).  But if your board rotation does not exactly match one of the available selections, you currently must measure the deviation and manually apply it using the SENS_BOARD_*_OFF parameters. The way I do that is to check whether the artificial horizon indicates level in roll/pitch after calibration when hovering (copter) or in level flight (plane). If not, I adjust the SENS_BOARD_*_OFF parameters until it is.

A user would not need the extra step if the artificial horizon showed level in cruise attitude (FW) or hover (MC) after calibration. This will not be the case if the board is not level when the aircraft is level.

tumbili, 2015-05-19:
Implemented via #2197

closed 2015-05-19 by LorenzMeier