PX4/PX4-Autopilot #2214: SBUS out should deliver the neutral position of servos

severinleuenberger, 2015-05-22:
currently the SBUS output begins at the moment, when the vehicle is armed. Before there is just no output at all. 
For SBUS Servos this works fine, And they are just quiet until armed. Thats ok. 

But for ESC/Motors this is a real issue. And for other things like parachute and LED-Drivers as well. 

Therefore; the correct handling on SBUS should be the same as for PWM-Out: Always the neutral Position (the trim from the calibration) should be delivered.

LorenzMeier, 2015-06-14:
We can do that for the next release. In general it would be great to have a developer for S.BUS, do you feel you can maybe find someone from the RC hobby community with some programming skills to pick this up?

LorenzMeier, 2015-07-03:
This is fixed in 615affd.

closed 2015-07-03 by LorenzMeier