PX4/PX4-Autopilot #2243: Create an enum on MAVLink so to define sensor ids

TSC21, 2015-05-27:
In a way of getting multi sub/pub working right for multiple rangefinder data sources, I think something like an enum must be defined so each range finder sensor as it's own ID on the Firmware (which will complete the `DISTANCE_SENSOR` msg fields).
I suggest something like `MAV_DISTANCE_SENSOR_ID` enum with the following ID's:
```
MAV_DISTANCE_SENSOR_ID_MB12XX  0
MAV_DISTANCE_SENSOR_ID_LL40LS  1
MAV_DISTANCE_SENSOR_ID_SF0X  2
MAV_DISTANCE_SENSOR_ID_TRONE  3
```
Anyone that adds a new driver for a range finder can then update this list. Also, if one has more than one sensor of each above, should use an ID greater or equal than 4 (obviously).
Anyone has a different opinion? @LorenzMeier, @mhkabir, @bansiesta, @Zefz, @jgoppert, @vooon

DrTon, 2015-05-27:
>Also, if one has more than one sensor of each above, should use an ID greater or equal than 4 (obviously).

This is the most inconsistent and bad part. Why not to simply set id via command line when starting driver?

TSC21, 2015-05-27:
> Why not to simply set id via command line when starting driver?

What do you mean? What about sensors which data is received by MAVLink serial connection? (which is actually possible right now)

TSC21, 2015-05-27:
We can always reserve an array of values for each type of sensor and impose a condition which doesn't let use the same id for pub/sub - this last one probably is a must.

vooon, 2015-05-27:
How about multiple LidarLite sensors? I think enum for id is a bad thing.

TSC21, 2015-05-27:
@vooon what about mavros side? What's the rul

vooon, 2015-05-27:
Mavros does mux/demux by id, so user can setup any id to any topic mapping.
FW side may also use id for mux ORB, if it really needed.

TSC21, 2015-05-27:
Still we are not setting up rules for sensor ids! And that was the purpose of this issue. As I said, without proper rules, probably we end up having the same id for many sensors. Could be interesting to list which range of ids are set to each type of sensor, I think.

DrTon, 2015-05-27:
>As I said, without proper rules, probably we end up having the same id for many sensors

No, just need to set ID by hand somewhere or do automatic enumeration, using type is wrong approach and must be avoided. Any rules will just confuse users later. If you want, feel free to add SENSOR_TYPE enum - no problem, but it will be NOT unique sensor ID and useless at this point, IMO.

TSC21, 2015-05-27:
Why do you say it's useless when we have some estimators depending on multiple sensor sources. In my opinion:
* set by hand - bad;
* set rules - better;
* set automatic enum - good;
* set automatic enum with a range for each sensor (like 1-5 -> mb12xx; 6-10 -> ll40ls) - best approach.

LorenzMeier, 2015-05-27:
An enum doesn't scale. The other sensors use a sensor ID by using the bus and address, which is unique.

What your are suggesting is a sensor type. Much better than the type is providing the properties like min range, max range, update rate and certainty. This is what the consuming algorithm needs to know.

TSC21, 2015-05-27:
> What your are suggesting is a sensor type. Much better than the type is providing the properties like min range, max range, update rate and certainty. This is what the consuming algorithm needs to know.

@LorenzMeier how do you suggest organize this properties in a ID list then? Thanks

LorenzMeier, 2015-05-27:
Not at all. I would offer them as individual fields.

TSC21, 2015-05-27:
But those are already fields of the msg (discarding update rate and certainty). What I'm trying to set here is a way of automatically set the id field of DISTANCE_SENSOR msg but as something with technical meaning, and not randomly set the id.

vooon, 2015-05-27:
@LorenzMeier hmm, its clever just use bus and address as id, but i'm not sure how to prevent conflicts?
At least this method allow auto-configure and do not change over time.

DrTon, 2015-05-28:
>At least this method allow auto-configure and do not change over time.

It doesn't prevent changing over time. Once you connect one more sensor of the same time somewhere - IDs will change. Actually why you need IDs at all? Estimator should know where sensor points and characteristics pointed by @LorenzMeier. If ID will be automatically assigned and change when reconnecting sensors - no problem. Just need to have unordered set of data topics here IMO. But for debugging and logging it's better to have unique IDs of course. Only robust way is to set by hands. Why do you think that it's bad? We already had a lot of problems with auto mag detection, the same here.

LorenzMeier, 2015-06-03:
Closing this for now - please send a PR against mavlink with a sensor message spec as discussed in case you want to proceed with this. Thanks all for the discussion!

closed 2015-06-03 by LorenzMeier

TSC21, 2015-06-03:
@LorenzMeier I want to proceed but I'm still not sure of how we are going to set the id's. And if we are to let them be set automatically, are they ordered on unordered and what sets that order.