PX4/PX4-Autopilot #2244: Level Calibration Not Reliable

jgoppert, 2015-05-27:
I am still seeing 0.3 g x acc on level ground after doing a level cal. Manually setting BOARD_ROT_Y fixed this for me.

kd0aij, 2015-05-27:
level cal worked ok for me on my f450 quad and Caipi_vtol

tumbili, 2015-05-27:
@jgoppert Could you put the vehicle level and check your euler angles. Can you then check if after the calibration the board offset values are more or less equal to these?

tumbili, 2015-05-27:
@jgoppert And what board rotation do you have?

LorenzMeier, 2015-06-03:
@jgoppert Can't replica this. Will close this unless we get more feedback.

jgoppert, 2015-06-03:
OK

edkoch, 2015-06-03:
I don't know if the issue I am seeing is the same as what has been reported here, but as I mentioned in the dev call the other day I am not able to keep a level calibration on my quad for very long. I am using the latest QGC and when I both calibrate the gyro and do the new level horizon calibration the level looks great as can be seen in the enclosed image, but after flying once or twice it gets out of wack and I have to recalibrate it again. I did not see this behavior with RC8.

In the enclosed diagrams I show a nice level view after a calibration and then also show the level after a couple of flights.

[image]
[image]
[image]

tumbili, 2015-06-03:
@edkoch How is your Pixhawk attached to the frame? Any chance that it could move?

tumbili, 2015-06-03:
@edkoch I don't want to start speculating but I remember to have had a similar issue with a Pixhawk with damaged sensors.

edkoch, 2015-06-03:
My platform is a 3DR X8 and I have not changed the Pixhawk mounting. It uses double sided foam tape so I guess it is a little pliable, but it should remain consistent from flight to flight. I have also been wondering if there could be a HW problem. I actually have a few Pixhawks on hand so I’ll swap it out and see if the problem persists.

-ed

tumbili, 2015-06-03:
@edkoch Yes, please do so!

edkoch, 2015-06-10:
I finally had a chance to swap out the pixhawk and after a couple of flights I don’t seem to be having the same problems that I was having. I’ll keep an eye on it, but perhaps I do have some bad HW.

-ed

LorenzMeier, 2015-06-12:
@jgoppert I'm closing this, please reopen if you continue to see this.

closed 2015-06-12 by LorenzMeier