PX4/PX4-Autopilot #2246: new flight mode "landing" for fixed-wing

severinleuenberger, 2015-05-28:
After nearly 6 Month of intense testing of "autoland" I see a great value for autolandings. But not always is a landing plannable by groudnstation. 

Therefore a lot of the functions from autoland could be implemented in a "landing" flight-mode. 
The main function would be: 
- hold the descend angle as declared by FW_LND_ANG
- hold the airspeed as in autoland as triggered by FW_AIRSPD_MIN, MAX, TRIM

Additionally if there is a range finder (laser distance sensor) and if it is activated by FW_LND_USETER =1
Then do flare as soon as reached the flare-altitude declared by FW_LND_FLALT
and do throttle handle as declared by FW_LND_TLALT
and do heading hold handle as declared by FW_LND_HHDIST

Then it is very important, that the user can interact the landing process by manually steering by RC. It should be the same input-behaviour as in STABILIZED, and not the limited action field as in ALTCTL/POSCTL (speaking of the new flight modes). 

I would call the new Mode for Landing simply "LAND" in PX4

I myself cannot do the code-change. But I would eagerly help you by defining the logic, helping on reviewing C++ Code and of course testing. 

Anyone interested on developing this?

dagar, 2016-03-11:
Would it be reasonable to require having a landing (and approach) waypoint defined? Then going into LAND mode would jump straight to the land portion of the mission.

DonLakeFlyer, 2016-03-11:
Why isn't that RTL with one or more MAV_CMD_DO_LAND_START items in your mission?

dagar, 2016-11-24:
@severinleuenberger we can do this if you help clarify a few details.

If you activated this mode would you want it to try and land at the home position, or start landing immediately?

Would it just do FW_LND_ANG immediately, or decrease altitude first?

closed 2018-02-22 by PX4BuildBot