PX4/PX4-Autopilot #2255: LL40LS PWM driver : excess CPU usage

mhkabir, 2015-05-29:
Reported by @jgoppert in #2096.
I have also confirmed the same issue, although on-bench.

TSC21, 2015-05-29:
Confirmed here also!

jgoppert, 2015-05-29:
@bansiesta could you take a look?

bansiesta, 2015-05-31:
How do I reproduce this?

This is **without ll40ls running**:

```
Processes: 20 total, 2 running, 18 sleeping
CPU usage: 34.84% tasks, 0.56% sched, 64.60% idle
Uptime: 167.388s total, 110.645s idle

 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE 
   0 Idle Task                  110644 64.601     0/    0   0 (  0)  READY 
   1 hpwork                       4516  2.735   780/ 1792 192 (192)  w:sig 
   2 lpwork                        754  0.402   580/ 1792  50 ( 50)  READY 
   3 init                         1437  0.000  1316/ 2992 100 (100)  w:sem 
   6 nshterm                         0  0.000   580/ 1496  70 ( 70)  w:sem 
 133 top                           113  2.172  1164/ 1696 100 (100)  RUN   
  62 dataman                        22  0.000   652/ 1792 100 (100)  w:sem 
  81 sensors_task                 4662  2.735  1300/ 1992 250 (250)  w:sem 
  83 gps                           295  0.160   716/ 1496 220 (220)  w:sig 
  85 commander                     798  0.402  2828/ 3392 215 (215)  w:sig 
  86 commander_low_prio            447  0.000   740/ 2592  50 ( 50)  w:sem 
  88 px4io                        2104  1.206   892/ 1792 240 (240)  w:sem 
  93 mavlink_if0                  1240  0.724  1836/ 2392 100 (100)  w:sig 
  95 mavlink_rcv_if0                12  0.000   796/ 2096 175 (175)  w:sem 
 107 sdlog2                        378  0.160  1868/ 2992  70 ( 70)  w:sig 
 112 ekf_att_pos_estimator       37496 22.847  3516/ 7496 215 (215)  w:sem 
 116 fw_att_control                867  0.482   900/ 1592 250 (250)  w:sem 
 121 fw_pos_control_l1              79  0.000   964/ 1592 250 (250)  w:sem 
 123 land_detector                 289  0.160   548/  992 100 (100)  w:sig 
 132 navigator                    1068  0.643   900/ 1696 120 (120)  w:sem 
```

Then **with ll40ls running**:

```
ll40ls start pwm
```

```
Processes: 20 total, 3 running, 17 sleeping
CPU usage: 35.04% tasks, 0.56% sched, 64.41% idle
Uptime: 323.445s total, 214.968s idle

 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE 
   0 Idle Task                  214967 64.405     0/    0   0 (  0)  READY 
   1 hpwork                       8873  2.793   780/ 1792 192 (192)  w:sig 
   2 lpwork                       1465  0.478   580/ 1792  50 ( 50)  READY 
   3 init                         1441  0.000  1316/ 2992 100 (100)  w:sem 
   6 nshterm                         0  0.000   580/ 1496  70 ( 70)  w:sem 
 137 top                           116  2.234  1164/ 1696 100 (100)  RUN   
  62 dataman                        22  0.000   652/ 1792 100 (100)  w:sem 
  81 sensors_task                 9018  2.713  1300/ 1992 250 (250)  w:sem 
  83 gps                           572  0.159   716/ 1496 220 (220)  w:sem 
  85 commander                    1455  0.399  2828/ 3392 215 (215)  w:sig 
  86 commander_low_prio            453  0.000   740/ 2592  50 ( 50)  w:sem 
  88 px4io                        4071  1.197   892/ 1792 240 (240)  w:sem 
  93 mavlink_if0                  2408  0.718  1836/ 2392 100 (100)  w:sig 
  95 mavlink_rcv_if0                23  0.000   796/ 2096 175 (175)  w:sem 
 107 sdlog2                        723  0.159  1868/ 2992  70 ( 70)  w:sig 
 112 ekf_att_pos_estimator       73052 22.905  3516/ 7496 215 (215)  w:sem 
 116 fw_att_control               1691  0.478   900/ 1592 250 (250)  w:sem 
 121 fw_pos_control_l1             139  0.000   964/ 1592 250 (250)  w:sem 
 123 land_detector                 564  0.159   548/  992 100 (100)  w:sig 
 132 navigator                    2087  0.638   900/ 1696 120 (120)  w:sem 
```

mhkabir, 2015-05-31:
@bansiesta Try covering the lens of the lidar, which'll cause errors.

bansiesta, 2015-05-31:
Covered, while it keeps resetting:

```
CPU usage: 34.98% tasks, 0.56% sched, 64.46% idle
Uptime: 38.165s total, 23.443s idle

 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(B
   0 Idle Task                   22636 64.490     0/    0   0 (  0)  READY 
   1 hpwork                       1007  2.866   704/ 1792 192 (192)  w:sem 
   2 lpwork                        154  0.398   580/ 1792  50 ( 50)  READY 
   3 init                         1438  0.000  1188/ 2992 100 (100)  w:sem 
   6 nshterm                         0  0.000   756/ 1496  70 ( 70)  w:sem 
 136 top                           625  2.070  1172/ 1696 100 (100)  RUN   
  62 dataman                        25  0.000   660/ 1792 100 (100)  w:sem 
  83 sensors_task                 1052  2.945  1300/ 1992 250 (250)  w:sem 
  85 gps                            70  0.159   708/ 1496 220 (220)  w:sig 
  87 commander                     249  0.398  2828/ 3392 215 (215)  w:sig 
  88 commander_low_prio            442  0.000   740/ 2592  50 ( 50)  w:sem 
  90 px4io                         467  1.194  1044/ 1792 240 (240)  w:sem 
 100 mavlink_if0                   268  0.716  1828/ 2392 100 (100)  w:sig 
 101 mavlink_rcv_if0                 3  0.000   796/ 2096 175 (175)  w:sem 
 109 sdlog2                         87  0.159  1868/ 2992  70 ( 70)  w:sig 
 114 ekf_att_pos_estimator        7715 22.850  3516/ 7496 215 (215)  w:sem 
 118 fw_att_control                182  0.477   884/ 1592 250 (250)  w:sem 
 123 fw_pos_control_l1              31  0.000   760/ 1592 250 (250)  w:sem 
 125 land_detector                  61  0.159   492/  992 100 (100)  w:sig 
 135 navigator                     226  0.557   924/ 1696 120 (120)  w:sem 
Idle Task: Lidar is down, reseting 218  0.556   924/ 1696 120 (120)  w:sem 
Idle Task: Lidar is down, resetingwn, reseting
```

mhkabir, 2015-05-31:
I'm getting around 50% total usage when it's reset-spamming. LPE + EKF
attitude estimator use around 23% together.

P.S Can you change the printf to a warnx for the reset, so that it doesn't
spam the USB console too :)

bansiesta, 2015-05-31:
Well, I can't reproduce it. You have to give me instructions, so I have the exact same setup respectively the same processes running.

bansiesta, 2015-06-03:
I'm closing this for now. Please re-open if needed.

jgoppert, 2015-06-03:
@bansiesta Please try against the estimator here(https://github.com/PX4/Firmware/pull/2096), you can enable it by setting LPE_ENABLED = 1.

Currently this estimator does not limit the data rate, maybe it is publishing faster than 10 Hz and using a lot of CPU?

bansiesta, 2015-06-04:
I flashed #2096, set LPE_ENABLED 1 and SYS_AUTOSTART 4011:

I see the following in top (independent of ll40ls reseting):

```
Processes: 21 total, 2 running, 19 sleeping
CPU usage: 49.53% tasks, 0.63% sched, 49.84% idle
Uptime: 70.355s total, 35.210s idle

 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE 
   0 Idle Task                   35210 49.842     0/    0   0 (  0)  READY 
   1 hpwork                       1928  2.764   796/ 1792 192 (192)  w:sig 
   2 lpwork                        311  0.473   580/ 1792  50 ( 50)  READY 
   3 init                         1444  0.000  1444/ 2992 100 (100)  w:sem 
   6 nshterm                         0  0.000   716/ 1496  70 ( 70)  w:sem 
 147 top                           323  1.974  1180/ 1696 100 (100)  RUN   
  63 dataman                        21  0.000   652/ 1792 100 (100)  w:sem 
  84 sensors_task                 2184  3.080  1300/ 1992 250 (250)  w:sem 
  86 gps                           131  0.157   724/ 1496 220 (220)  w:sig 
  88 commander                     443  0.394  2860/ 3392 215 (215)  w:sig 
  89 commander_low_prio            458  0.000   740/ 2592  50 ( 50)  w:sem 
  91 px4io                        2242  3.080   948/ 1792 240 (240)  w:sem 
 101 mavlink_if0                   524  0.789  1828/ 2392 100 (100)  w:sig 
 103 mavlink_rcv_if0                 5  0.000   796/ 2096 175 (175)  w:sem 
 110 sdlog2                        165  0.236  1868/ 2992  70 ( 70)  w:sig 
 125 attitude_estimator_ekf      12854 19.036  6164/ 7696 250 (250)  w:sem 
 136 lpe                          7453 12.006  5572/ 6096 245 (245)  w:sem 
 138 mc_att_control               1694  2.527  1020/ 1496 250 (250)  w:sem 
 140 mc_pos_control               1109  1.579  1196/ 1496 250 (250)  w:sem 
 143 land_detector                 134  0.157   468/  992 100 (100)  w:sig 
 145 navigator                     779  1.263   836/ 1696 250 (250)  w:sem
```

jgoppert, 2015-06-07:
I can no longer reproduce the cpu usage either. Can anyone else verify this is fixed for the LPE branch now? @mhkabir, @TSC21

closed 2015-06-12 by bansiesta